FreeOpcUa/freeopcua #108: Cannot write to the node of custom server

destogl, 2015-01-09:
Hello,

I have custom server based on Unified Automation SDK and I can not write new value to the node. Writing is possible with UaExpert. Also the example client can write to the example server (provided in this repo).

I tried to debug it with wireshark but I'm not sure for what should I search. I saw there is a quite a difference in data which are sent. Do you have any recommendation how should I proceed?

oroulet, 2015-01-10:
Which SDK are you using? C++ or .net?
Can you send me a log which is failing with FreeOpcUa? And the terminal
output?

destogl, 2015-01-10:
I am not sure which SDK because our project partners have a licence for this SDK, but I think is .NET.

I tried to extract the Writing event in Wireshark data so I hope it will be easier for you to check it. The variable is writeable (UaExpert says that) and in booth cases I am writing value 512 to variable ns=2; s=3001.Mode.

So here are the logs:

[Terminal output](http://marvin.kset.org/~denis/Posao/term_output.log)

[Wireshark log FreeeOpcUa client](http://marvin.kset.org/~denis/Posao/FreeOpcUa_client.pcapng) - only write event

[Wireshark log UaExpert](http://marvin.kset.org/~denis/Posao/UaExpert.pcapng) - write event begins at 0.0 (and I am not sure when it ends because of keepalive stuff)

oroulet, 2015-01-10:
I had a short look at it. looks like the only difference is that we set the
source timestampt while uaexpert does not. I think I remember something
similar with a plc.
I suppose that you write a variant using SetValue. Try to create a
DataValue object and do not set the variant, but do not set the
sourcetimestamp then write it using SetValue.
Let me know if it works
If it does we will remove this timestampt stuff, it seems many servers do
not support it

oroulet, 2015-01-10:
SET the variant and do NOT set the timestamp... I mean

destogl, 2015-01-10:
It works, and I created Pull-request with a quick hack (#109)

closed 2015-01-10 by destogl

oroulet, 2015-01-10:
I am unsure what is the correct solution here. I think I will just remove
the second argument. it wil simplify the api, and if someone really want to
set the source timestamp they can always use DataValue

destogl, 2015-01-12:
I am not sure how protocol is defined. I would go for a solution which is more according to the standard....