FreeOpcUa/freeopcua #116: Client crashes when connected and disconnected more than 2 times in the row

fvacek, 2015-02-25:
Problem is that `StopRequest` is not set bac to false when KeepAlive thread is started again. Patch attached.

```diff
diff --git a/src/client/client.cpp b/src/client/client.cpp
index 399c31b..ab355c9 100644
--- a/src/client/client.cpp
+++ b/src/client/client.cpp
@@ -35,6 +35,7 @@ namespace OpcUa
     NodeToRead = node;
     Period = period;
     Running = true;
+    StopRequest = false;
     Thread = std::thread([this] { this->Run(); });
   }
 
@@ -44,8 +45,8 @@ namespace OpcUa
     if (Debug)  { std::cout << "KeepAliveThread | Starting." << std::endl; }
     while ( ! StopRequest )
     {
-      std::unique_lock<std::mutex> lock(Mutex);
       if (Debug)  { std::cout << "KeepAliveThread | Sleeping for: " << (int64_t) ( Period * 0.7 )<< std::endl; }
+      std::unique_lock<std::mutex> lock(Mutex);
       std::cv_status status = Condition.wait_for(lock, std::chrono::milliseconds( (int64_t) ( Period * 0.7) )); 
       if (status == std::cv_status::no_timeout ) 
       {
```

oroulet, 2015-02-25:
thanks, this is really something I did not think of! Can't you just push this in you branmch and create a pull request?

fvacek, 2015-02-25:
Actualy I like to, but I have pending pull request https://github.com/FreeOpcUa/freeopcua/pull/113 and it bloks me from creating more patches without forking again.

oroulet, 2015-02-25:
actually the usuall way to do it is to create a new branch and push from that one to github (then creating a new brnch github)
git checkout -b bugfix
git reset --hard "head of master on github"
make your changes
git push to github
create new pull request

fvacek, 2015-02-26:
https://github.com/FreeOpcUa/freeopcua/pull/117

Anyway, I still wonder why are you using rather big C++ functions inlining?

closed 2015-03-05 by fvacek