FreeOpcUa/freeopcua #123: DataDeserializer::Deserialize<Variant>(Variant& var) doesn't support VariantType::EXTENSION_OBJECT

fvacek, 2015-03-05:
DataDeserializer::Deserialize<Variant>(Variant& var) doesn't support VariantType::EXTENSION_OBJECT

this leads to communication errors because extension oject data are missinterpreted by deserializer. Even if the EXTENSION_OBJECT is not supported, its data shall be at least skipped with some warning. In my case, the deserializer is expecting  1538353707 bytes of data instead of 184.

oroulet, 2015-03-05:
> DataDeserializer::Deserialize(Variant& var) doesn't support

Ja support for ExtensionObjects should be added. there are many cases where
extensionObjects are used. sending structs for examples
https://github.com/FreeOpcUa/freeopcua/issues/52

fvacek, 2015-03-05:
Shouldn't we at least throw an exception about unsupported variant type deserialization? It is much more descriptive than "expecting 2345667 bytes of data".                                                                                                                                                                                                                                                                                                                                         Sent from my BlackBerry 10 smartphone.                                                                                                                                                                                                                From: ORDSent: Thursday, March 5, 2015 4:07 PMTo: FreeOpcUa/freeopcuaReply To: FreeOpcUa/freeopcuaCc: Fanda VacekSubject: Re: [freeopcua] DataDeserializer::Deserialize<Variant>(Variant& var) doesn't support VariantType::EXTENSION_OBJECT (#123)Ja support for ExtensionObjects should be added. there are many cases where
extensionObjects are used. sending structs for examples
https://github.com/FreeOpcUa/freeopcua/issues/52

oroulet, 2015-03-05:
Yes of course!

oroulet, 2015-03-06:
but implementing ExtrensionObject should be really easy,
deserialization code should already be written it is used many other
places (like in request headers), so it is just a matter of adding
support for it in variant.

fvacek, 2015-03-06:
Ok, I'll take look at it.

Fanda

informatikr, 2016-12-19:
I think this is fixed by #214.

closed 2016-12-19 by oroulet