FreeOpcUa/freeopcua #137: Easy: Port as many structs as possible to auto-generated code

oroulet, 2015-05-21:
from older epost:
This push request enabled auto-generation #129 (alreadt merged)

auto-generated files end with _auto

Per default most of
the auto-generated code is commented out. As an example I removed all
code in attribute.h and replaced it with auto-generated code. But this
should be done for all simple structures and custom enum. but NOT for
more complicated stuff like NodeAttributes.

We also have a lot of custom enums that should be removed and use
ObjectId or autogenerated enums.

If someone is interrested take a look at schemas/generate_protocol,py
and try to uncomment the structs you are interested in. then run
generate_protocol .py. You will then need to remove hand written code.
The python code is written very quickly from the parser I made for
python but the generated c++ code should be OK

The autogeneated structs are sometimes different from those
handwritten. Then you need to do some small renaming but it should be easy

destogl, 2015-06-02:
TODOs after full porting:

* Remove casting to ByteString

Issues:

* ActivateSessionParamters - ExtensionObject

oroulet, 2015-06-02:
ActivatesessionParameters is not an extension object so this not an issue.
UserIdentityToken is an extensionObject. You can just leave the manual implementation .
Is there an issue here?

destogl, 2015-06-02:
Issue is that autogenerated structure ActivateSessionParameters has UserIdentityToken defined as ExtensionObject. Can this be somehow resolved to use manual implementation of UserIdentityToken?

destogl, 2015-06-02:
Never mind, just started to extend generator for this cases

oroulet, 2015-06-02:
>> Can this be somehow resolved to use manual implementation of
UserIdentityToken?
Yes I think this is a good short term solution. and I advice you to stay
there for now.

If you want to go further: autogenerating extended objects has been
discussed a lot earlier. There are many different solutions. I think the
easiest is to have a generic extended object which can pack any ua
structure. Thus UserIdentityToken would be a normal structure that we need
to pack in extension object before sending.
be carefull definition of extended oebjct is wrong in xml. I generate  a
corrected version in python-opcua but I am not sure it is up to date in C++

destogl, 2015-06-03:
Why not to do this ExtensionObject as a basis structures for other objects. I think they are meant to use like this. And this is relatively easy to do with C++ structures. IMHO this provides higher robustness.

oroulet, 2015-06-04:
I remeber trying both with inheritance and without. The issue is what to do
when deserializing an extension object.

We can have a very complicated deserialization method for extension object
that check typeId and tries to deserialize but this will never work for all
types since one ca have user types.

So I think we need generic ExtensionObject class anyway. Which basically
are a typeid and a bytestring.
Then let user serialize deserialize their own type to and from bytestring.

If we use inheritance serializing is easy
MyObject >> stream
But deserializing is more trickyp
Stream >> ExtensionObject
ifExtensionObjectTypeID == MyObjectId
{
Extensionobject.ByteString >> MyObject
}

But if you have a good idea. Jus try it :-)

>