FreeOpcUa/freeopcua #139: Client authentication

tomkcook, 2015-05-22:
Quick question:  Does the FreeOpcUa library support client authentication (with username/password)?

If not, is there any chance you could give a quick pointer to how it might be implemented?

oroulet, 2015-05-22:
look at the example file. should be a line there to show how to set user
name and passwrod

tomkcook, 2015-05-25:
Do you mean using an endpoint url like this:

    opc.tcp://user:password@server:4000/

?  If so, it doesn't work for me; I get the message `No supported endpoints found on server`.  I suspect this is because of the hard-coded security policy, `http://opcfoundation.org/UA/SecurityPolicy#None`.

If I modify my PLC's configuration to allow anonymous access then it works, but that rather defeats the purpose of client authentication!

tomkcook, 2015-05-25:
I'm guessing that the security policies other than None are not supported?

oroulet, 2015-05-25:
I am sure someone is using user authentication so it has worked in one or
another  configuration....But i never had use for it myself. It should not
be hard to fix. Just fill up the authentication structure as the server
wants

tomkcook, 2015-05-25:
Any pointers towards how to figure out what it wants?  I'm at a bit of a loss.

oroulet, 2015-05-25:
I am not really sure about what you know or not, but let's try:
look at the connect method of the Client class. that method has become far
too long over time and should be split!!!!!!!! but at the end what it does
is call 3 methods from BinaryClient class
->OpenSecureChannel
->CreateSession
->ActivateSession
each of this methods take a struct as argument. the trick is to find out
what your server wants to connect to it.
You can read the long and complicated opc-ua spec or simpoly fire up a
client like uaexpert, fill up connection information and monitore session
with wireshark. Wireshark has support for opcua protocol,so it is just a
mater of reading what it sends to the server and write the same in your
client code.

oroulet, 2015-09-18:
just for information I found out the implementation of password i broken in python-opcua so it also probably broken here. The password need to be encrypted before sending...

joaopmrod, 2016-02-18:
I'm trying to connect using this: opc.tcp://admin:password@localhost:4096 and  I get this error:

showing error:  name 'uri' is not defined 1.

If I enable anonymous access I can do it.