FreeOpcUa/freeopcua #142: Unclear License

smyrman, 2015-05-31:
The root level COPYING file says the software is released as LGPL (Lesar GNU General Public License). So does python/COPYING. However, python/LICENSE says GPL, and so does several header and source files!

Legally, I can only interpret this as if the whole software is GPL.

Here is an abstract from git grep " GPL".
include/opc/common/addons_core/config_file.h:/// @license GNU GPL
include/opc/common/addons_core/config_file.h:/// Distributed under the GNU GPL License
include/opc/ua/client/addon.h:/// @license GNU GPL
include/opc/ua/client/addon.h:/// Distributed under the GNU GPL License
include/opc/ua/server/addons/address_space.h:/// @license GNU GPL
include/opc/ua/server/addons/address_space.h:/// Distributed under the GNU GPL License
include/opc/ua/server/addons/standard_address_space.h:/// @license GNU GPL
include/opc/ua/server/addons/standard_address_space.h:/// Distributed under the GNU GPL License
include/opc/ua/server/addons/subscription_service.h:/// @license GNU GPL
include/opc/ua/server/addons/subscription_service.h:/// Distributed under the GNU GPL License
include/opc/ua/server/addons/xml_ns.h:/// @license GNU GPL
include/opc/ua/server/addons/xml_ns.h:/// Distributed under the GNU GPL License
include/opc/ua/server/standard_address_space.h:/// @license GNU GPL
include/opc/ua/server/standard_address_space.h:/// Distributed under the GNU GPL License
python/configure.ac:# Distributed under the GNU GPL License
python/src/py_opcua_enums.cpp:/// @license GNU GPL
python/src/py_opcua_enums.cpp:/// Distributed under the GNU GPL License
python/src/py_opcua_enums.h:/// @license GNU GPL
python/src/py_opcua_enums.h:/// Distributed under the GNU GPL License
python/src/py_opcua_helpers.h:/// @license GNU GPL
python/src/py_opcua_helpers.h:/// Distributed under the GNU GPL License
python/src/py_opcua_module.cpp:/// @license GNU GPL
python/src/py_opcua_module.cpp:/// Distributed under the GNU GPL License
python/src/py_opcua_subscriptionclient.cpp:/// @license GNU GPL
python/src/py_opcua_subscriptionclient.cpp:/// Distributed under the GNU GPL License
python/src/py_opcua_subscriptionclient.h:/// @license GNU GPL
python/src/py_opcua_subscriptionclient.h:/// Distributed under the GNU GPL License
python/src/py_opcua_variant.h:/// @license GNU GPL
python/src/py_opcua_variant.h:/// Distributed under the GNU GPL License
src/client/binary_client_addon.cpp:/// @license GNU GPL
src/client/binary_client_addon.cpp:/// Distributed under the GNU GPL License
src/clientapp/opcua_main.cpp:/// @license GNU GPL
src/clientapp/opcua_main.cpp:/// Distributed under the GNU GPL License
src/clientapp/opcua_options.cpp:/// @license GNU GPL
src/clientapp/opcua_options.cpp:/// Distributed under the GNU GPL License
src/clientapp/opcua_options.h:/// @license GNU GPL
src/clientapp/opcua_options.h:/// Distributed under the GNU GPL License
src/core/common/addons_core/config_file.cpp:/// @license GNU GPL
src/core/common/addons_core/config_file.cpp:/// Distributed under the GNU GPL License
src/examples/example_client.cpp:/// @license GNU GPL
src/examples/example_client.cpp:/// Distributed under the GNU GPL License
src/examples/example_server.cpp:/// @license GNU GPL
src/examples/example_server.cpp:/// Distributed under the GNU GPL License
src/server/address_space_addon.h:/// @license GNU GPL
src/server/address_space_addon.h:/// Distributed under the GNU GPL License
src/server/endpoints_parameters.cpp:/// @license GNU GPL
src/server/endpoints_parameters.cpp:/// Distributed under the GNU GPL License
src/server/endpoints_parameters.h:/// @license GNU GPL
src/server/endpoints_parameters.h:/// Distributed under the GNU GPL License
src/server/standard_address_space_addon.cpp:/// @license GNU GPL
src/server/standard_address_space_addon.cpp:/// Distributed under the GNU GPL License
src/server/subscription_service_addon.cpp:/// @license GNU GPL
src/server/subscription_service_addon.cpp:/// Distributed under the GNU GPL License
src/server/xml_address_space_addon.cpp:/// @license GNU GPL
src/server/xml_address_space_addon.cpp:/// Distributed under the GNU GPL License
src/server/xml_address_space_loader.cpp:/// @license GNU GPL
src/server/xml_address_space_loader.cpp:/// Distributed under the GNU GPL License
src/server/xml_address_space_loader.h:/// @license GNU GPL
src/server/xml_address_space_loader.h:/// Distributed under the GNU GPL License
src/server/xml_processor.h:/// @license GNU GPL
src/server/xml_processor.h:/// Distributed under the GNU GPL License
src/serverapp/daemon.cpp:/// @license GNU GPL
src/serverapp/daemon.cpp:/// Distributed under the GNU GPL License
src/serverapp/daemon.h:/// @license GNU GPL
src/serverapp/daemon.h:/// Distributed under the GNU GPL License
src/serverapp/daemon_lin.cpp:/// @license GNU GPL
src/serverapp/daemon_lin.cpp:/// Distributed under the GNU GPL License
src/serverapp/daemon_win.cpp:/// @license GNU GPL
src/serverapp/daemon_win.cpp:/// Distributed under the GNU GPL License
src/serverapp/server_main.cpp:/// @license GNU GPL
src/serverapp/server_main.cpp:/// Distributed under the GNU GPL License
src/serverapp/server_options.cpp:/// @license GNU GPL
src/serverapp/server_options.cpp:/// Distributed under the GNU GPL License
src/serverapp/server_options.h:/// @license GNU GPL
src/serverapp/server_options.h:/// Distributed under the GNU GPL License
tests/gmock/gtest/m4/acx_pthread.m4:dnl @license GPLWithACException
tests/gtest/m4/acx_pthread.m4:dnl @license GPLWithACException
tests/protocol/utils.cpp:/// @license GNU GPL
tests/protocol/utils.cpp:/// Distributed under the GNU GPL License
tests/server/address_space_registry_test.h:/// @license GNU GPL
tests/server/address_space_registry_test.h:/// Distributed under the GNU GPL License
tests/server/builtin_server_test.h:/// @license GNU GPL
tests/server/builtin_server_test.h:/// Distributed under the GNU GPL License
tests/server/endpoints_services_test.h:/// @license GNU GPL
tests/server/endpoints_services_test.h:/// Distributed under the GNU GPL License
tests/server/opcua_protocol_addon_test.h:/// @license GNU GPL
tests/server/opcua_protocol_addon_test.h:/// Distributed under the GNU GPL License
tests/server/services_registry_test.h:/// @license GNU GPL
tests/server/services_registry_test.h:/// Distributed under the GNU GPL License
tests/server/standard_namespace_test.h:/// @license GNU GPL
tests/server/standard_namespace_test.h:/// Distributed under the GNU GPL License
tests/server/xml_address_space_addon_ut.cpp:/// @license GNU GPL
tests/server/xml_address_space_addon_ut.cpp:/// Distributed under the GNU GPL License

oroulet, 2015-05-31:
Good point. this is an error. everything should be LGPL. The question now
is how ti fix that ..... I see that almost all impacted files are
completely written by me or Alexander. So this is easy to fix.  however
some files have small patches from others...... I do not know how to solve
that

destogl, 2015-05-31:
If there is any patch from me (didn't checked) you can licence it under LGPL.

RavilN, 2015-06-03:
I don't think that if somebody checks in some file, he can change licensing terms by doing that.
As I understood, the license is LGPL, and it is going to be that way forever, and cannot be changed just by checking-in something.
Please correct me if I wrong.

oroulet, 2015-06-03:
The project on gihub is flagged as LGPL, the COPYING file states LGPL, the
README says LGPL and the website says LGPL and there are many emails where
the original author and me state that the software is LGPL so I think it is
not problem to fix the few header strings where it written GPL instead of
LGPL.

I someone send a patch to fix those strings I will merge it

closed 2015-09-18 by oroulet