FreeOpcUa/freeopcua #150: Autogeneration of operators and tests

destogl, 2015-06-05:
During autogeneration spree (PR: #147 and additional comments #137) the need for generation of structure operators (e.g. '==') and possibility for autogeneration is diagnosed. So this is something what is IMHO valuable to look at.

Or do we really need test for autogenerated structure? Maybe would be more logical (and enough) to have test for autogeneration script.

What do you think?

oroulet, 2015-06-05:
I fo not know if there is a correct answer here. I do not think we need to
autogenerate tests from xml spec.  but we should have some tests to check
that the produced autogenerated code is correct. We should select a few
complicated structs and check that serialization and deserailization
happens as expected. For example there are auto-generated binary tests ala

        if ((data.Encoding) & (1>>(0))) *this >> data.Body;
    }

And add tests at high level that use those structs. For example we have
currently no tests for method call...

destogl, 2015-06-05:
Yes I agree, to have few tests for serialisation and deserialisation and than write the tests for functionalities of the library (e.g. method call). But here I have one question: How to test functionality separately for the client and server? Is there anything like this provided by OPC UA fundation?

oroulet, 2015-06-05:
We have nothing, but the opc foundation has a test suite which is only
available to members. I think someone on the list said he could tests
things for us if we wanted. you can try to search the list.
To test the client I thought about creating a file test_client.cpp that
connect to an arbitrary server and that run all possible tests to access
compatibilty with that particular server.
 I never had time to do it. although using python and unittest it should be
very easy..