FreeOpcUa/freeopcua #155: Subscriptions and constness

tomkcook, 2015-06-25:
Is there any reason why the member functions of SubscriptionHandler are declared const?  It makes it rather a pain to actually do anything with the updates.  For instance, my SubscriptionHandler looks something like this:

    class MySubscriptions : public SubscriptionHandler {
        std::map<uint32_t, MyDataObjectType> subscribedNodes;
        Subscription subscription;

    public:
        void Subscribe(Node& n, MyDataObjectType o) {
            subscribedNodes[n.GetId().ServerIndex] = o;
            subscription.SubscribeDataChange(n);
        }

        void DataChange(uint32_t handle, const OpcUa::Node& node, const OpcUa::Variant& val, OpcUa::AttributeId attribute) const override {
            // What to do here?
        }
    };

Because the method is declared const, it's not possible to look up a value in the map as `MyDataObjectType o = subscribedNodes[node.GetId().ServerIndex]`.  Even when you do manage to get a node out of the map, it has to be a const reference because the method is declared const, so you can't call any non-const method on it.

Is there any real gain from declaring these methods const?  Would it be a problem to change to declaration to non-const?

oroulet, 2015-06-25:
No i do no think we have a good reason

destogl, 2015-09-16:
Solved in #159

closed 2015-09-16 by destogl