FreeOpcUa/freeopcua #161: Node identifiers & protocol efficiency

tomkcook, 2015-09-07:
We are currently using the code I added a while ago to write multiple nodes simultaneously to write 24 variables in a single network packet.  This results in packets of 1397 bytes, approaching the Ethernet MTU.

The main problem here seems to be that every write packet contains the node name and path as a string.  Is there some way to avoid this?

oroulet, 2015-09-07:
Very good question, I always though that this protocol is far too verbose.
But looking at the spec I do not see any way avoid sending the node id for
every attribute one want to write.
The spec is now available to everybody who register to opc website, so you
can have a look. Part 4 is the one. and look for write request.

tomkcook, 2015-09-15:
AFAICT, the right way to do this is to use the `RegisterNodes` request (OPC UA spec part four, section 5.8.5).  This takes an array of `NodeId`s as a parameter and returns an array of `NodeId`s.  The server is allowed to just copy the input to the output, but it can also provide a temporary `NodeId` which it is able to process more efficiently - eg. a numeric identifier instead of a string.  Obviously this is much more efficient, both in terms of protocol and processing.

I can't see anywhere that the `RegisterNodes` request is exposed in freeopcua at present.  I'm not sure what the best way of doing it would be.  The options I can think of are, declared in the `ViewServices` class and implemented in `BinaryClient`:

```
std::vector<OpcUa::NodeId> RegisterNodes(std::vector<OpcUa::NodeId>& ids);
```

This would just take a list of `NodeId`s and return the `NodeId`s returned by the server; it'd be up to the application to make sure the more efficient `NodeId`s got used in future requests.  Alternatively:

```
void RegisterNodes( std::vector<OpcUa::Node>& nodes );
```

This would take a list of `Node`s, query the server for better `NodeId`s to use to refer to them and then update the `Node` objects with the resulting `NodeId`s.  If the application then went on to use these Node objects, the better `NodeId`s would get used implicitly.

I'm leaning towards the second option, but I'd be keen to hear the thoughts of others.  Actually, perhaps the best thing would be to implement the first option in `ViewServices` and then the second option in `ServerOperations`?

tomkcook, 2015-09-15:
I've created a prototype implementation in #164.  I don't have a server to hand to test it against.

It's valid for a server to do nothing to both these requests, with RegisterNodes just returning the same list of NodeIds it was passed, so it might take some care to check it's working right.

oroulet, 2015-09-17:
merged with 166

closed 2015-09-17 by oroulet