FreeOpcUa/freeopcua #162: Error when I include some headers

BartP88, 2015-09-07:
Hey,

I'm using embarcadero X8 and when I include the client.h or any other header-file (from the example), it gives the following error:

[bcc32 Error] datetime.h(36): Non-virtual function 'DateTime::DateTime(const DateTime &)declared pure.
And I get another 50 errors with this. So did I forget something to include?

destogl, 2015-09-07:
Hello BartP88,

I just checked in my code and you didn't forget anything. It is enough to include just client.h. The problem could be in compiler options, are you using some special options?

BartP88, 2015-09-07:
I never changed anything about the compiler options. So it's just the basic compiler I guess.. Anyways when I add the header files like the example it complains about an error found in the file: datetime.h.
And its complaining about this line:    DateTime(const DateTime&) = default;

oroulet, 2015-09-07:
sound like a compiler issue. Embarcadero have their own compiler I guess?
we tested with VS, gcc and clang, there are probably issues with other
compilers. Patch are welcome if they are not workaround for crappy
compilers..

BartP88, 2015-09-07:
Any suggestions I should change about the compiler options to get it work?

destogl, 2015-09-07:
I am compiling everything with my project and linking openOpcUa library with my stuff. And than of course you should be able to include "client.h" without any problems and use client with it... Which C++ standard does your compile use?

BartP88, 2015-09-07:
I just made a new project for testing, added the 3 libraries only, and it's giving that error.

I'm using Embarcadero C++ builder XE8

destogl, 2015-09-09:
Which compiler are you using?

BartP88, 2015-09-09:
It seems I'm using a Delphi Compiler:  http://docwiki.embarcadero.com/RADStudio/XE8/en/Compiler_Versions

destogl, 2015-09-09:
Can you check the compatibility of that compile to standard compiles like gcc? Because I think that your compiler like this library very much :D

Do you have to use this compiler?
Which OS are you using?

BartP88, 2015-09-10:
I'm using Windows 7, and yes, I kinda have to.

I've added a screenshot about the options I can pick in the compiler. Maybe you can tell me or help me out with what I should change :)

[image]

destogl, 2015-09-15:
I have really no idea. maybe Force C++ compile?

destogl, 2015-09-15:
Is your compiler compatible with C++11?

BartP88, 2015-09-15:
I have no clue. But I asked the same question on stackoverflow, and people are saying it could be because my project is in 32bit instead of 64bit

destogl, 2015-09-19:
Did you try to compile it as 32bit?

destogl, 2015-10-02:
Hello @BartP88 did you find solution?

destogl, 2016-05-02:
Ping?