FreeOpcUa/freeopcua #163: make install is missing headers

basyskom-meerkoetter, 2015-09-11:
I build and install by doing 

./configure --prefix=/usr/local && make && sudo make install

The resulting install in /usr/local/include is missing headers. The most obvious example is global.h which is expected to be found in opc/ua/. global.h is referenced by other freeopcua headers which get installed. The result is that i can't compile a program against the install in /usr/local. The same thing works when compiling against the headers in the source tree.

I can "fix" my install by manually copying all headers to /usr/local/include.

oroulet, 2015-09-11:
Can you send a patch ! I currently do not have time to work on c++ stuff

basyskom-meerkoetter, 2015-09-17:
Fixed now

closed 2015-09-17 by basyskom-meerkoetter