FreeOpcUa/freeopcua #171: BinaryClient callback thread is not stopped when client disconnected.

fvacek, 2015-12-02:
This happens when subscription is created only.

__steps to reproduce:__
* run example_client with debug enabled
* check log when app finished, there is no lines:
```
binary_client| Stopping callback thread.
binary_client| CallbackThread : stopping.
binary_client| Joining service thread.
binary_client| CallbackThread : exited.
```
in output
* comment subscription part of demo
* run app again
* check log, above lines are present now

__reason:__
cyclic reference Server shared_ptr object is made, when subscription object is created. 
Server remains referenced form cache fields:
* BinaryClient::PublishCallbacks
* Subscription::AttributeValueMap

I'm not sure if I can make correct patch since the bug is tied to freeopcua design. I suggest at least clear both caches on disconnect.

oroulet, 2015-12-03:
yes this shared_prt stuff is a mess. The client code is not very complicated so if you have a better design idea, just let us know. Now I have done so much Python the last months that I am afraid to touch any C++ code. Btw If you want I can make you a member of freeopcua group so you can also merge patch or change things directly when necessary

fvacek, 2015-12-03:
I suggest to delete both maps on disconnect as a quick-fix. 

One of possible fix can be:
* to use weak_ptr in Node and Subscription classes, simply never use shared_ptr in object returned by API. This way we can be sure that nobody can hold strong reference to Server object and that it can be properly destroyed.

I like to be a reviewer and help other members of the project with the review process. But I don't feel myself to be as deep in freeopcua internals to be an approver.