FreeOpcUa/freeopcua #172: Build under windows fails

cedricboudinet, 2015-12-14:
Hi

I tried to compile freeopcu under windows 7 with cmake 3.4.1 and Visual Studio 2013.

I launch : 
mkdir build.msvc12.cmd
cd build.msvc12.cmd
cmake .. -DBOOST_ROOT=C:\boost_1_59_0 -DBOOST_LIBRARYDIR=c:\boost_1_59_0\lib32-msvc-12.0
cmake --build .

It fails on that step :
C:\Program Files\Microsoft Visual Studio 12.0\VC\bin\link.exe /ERRORREPORT:QU
EUE /OUT:"F:\Boudinet\Apps\freeopcua\build.msvc12.cmd\bin\Debug\opcuacore.dll
" /INCREMENTAL /NOLOGO kernel32.lib user32.lib gdi32.lib winspool.lib shell32
.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib Wininet.lib ws
ock32.lib lib\Debug\opcuaprotocol.lib Wininet.lib wsock32.lib /MANIFEST /MANI
FESTUAC:"level='asInvoker' uiAccess='false'" /manifest:embed /DEBUG /PDB:"F:/
Boudinet/Apps/freeopcua/build.msvc12.cmd/bin/Debug/opcuacore.pdb" /SUBSYSTEM:
CONSOLE /TLBID:1 /DYNAMICBASE /NXCOMPAT /IMPLIB:"F:/Boudinet/Apps/freeopcua/b
uild.msvc12.cmd/lib/Debug/opcuacore.lib" /MACHINE:X86 /SAFESEH  /machine:X86

with the message:
LINK : fatal error LNK1104: cannot open file 'lib\Debug\opcuaprotocol.lib' [F:\
Boudinet\Apps\freeopcua\build.msvc12.cmd\opcuacore.vcxproj]

Library bin\Debug\opcuacore.dll exists. It seems that it has been built in place of lib\Debug\opcuaprotocol.lib.
How to force msvc to generate all the .lib files automatically ?

Zorze93, 2015-12-15:
Hello,
I have exactly the same problem (exception made for the version of Visual Studio, I've the 2015 one), if someone could help I'd be grateful.

KnightChang17, 2015-12-27:
Hello,
I have the same problem, please help

MeldirAran, 2016-03-28:
I have the same problem, when I copy opcuaprotocol.lib to \lib directory and then compile opcuacore, I got 11 unresolved externals like:
error LNK2019: unresolved external symbol "class boost::system::error_category const & __cdecl boos
t::system::system_category(void)" (?system_category@system@boost@@YAABVerror_category@12@XZ) referenced in function "vo
id __cdecl boost::system::`dynamic initializer for 'native_ecat''(void)" (??__Enative_ecat@system@boost@@YAXXZ)

marcassin22, 2016-06-10:
Hi,
Same problem for me on Visual Studio 2013. Can I get any hints on how to change the configuration type?

tbeu, 2016-09-16:
> Can I get any hints on how to change the configuration type?

Do not select BUILD_SHARED_LIBS in CMake.

tbeu, 2016-09-22:
The real problem with the missing import libraries when building shared libraries using MSVC is that there are no exported symbols at all in the shared libraries. For that reason there are no import libraries built. Thus it really is bug in the Win build configuration.

destogl, 2016-09-26:
@tbeu can you make PR with correct configuration?

tbeu, 2016-09-26:
Sorry, I am not a CMake expert.