FreeOpcUa/freeopcua #174: getting through http proxy

cedricboudinet, 2015-12-15:
Hello
I am running a client in a network where ports are filtered and http connections to www have to pass through a proxy.
Is that possible to make the client use the proxy in order to access a remote opcua server ?

oroulet, 2015-12-15:
I have absolutely no idea. I have never used a proxy

cedricboudinet, 2015-12-16:
Ok thanks for the answer.
I've made a simple test of setting http_proxy and trying to connect to a server on a remote network and it hangs.

When I set up my endpoint (on the server or client side), is there any impact of the scheme of the uri? The default is opc.tcp://. I tried to change to htpp:// (second protocol supported by OPC UA) and I managed to connect a client talking in http:// to my server listening in opc.tcp://.
Does it mean that whatever the scheme is, the protocol is always binary (opc.tcp) ?

treww, 2015-12-16:
Hello.

It is not possible to pass opc.tcp through http proxy. It is binary
protocol.
There are soap bindings. they works over http and may be proxied natively.

best regards,
Alexander.

oroulet, 2015-12-16:
we only support binary protocol. adding support for soap is fully possiblem, it will require some work though...

cedricboudinet, 2015-12-16:
Ok
When you talk about soap bindings, you mean external tools? Such as using a tunnel (e.g. https://pagekite.net/wiki/Howto/OpcUAOverPageKite/) ?

cedricboudinet, 2015-12-19:
Or do you mean opcua-wsdl ?
Does it work alongside freeopcua?