FreeOpcUa/freeopcua #175: C++ Client cannot deserialize Python 0.10.3 FreeOpcUa server Xml Schema node?

Harha, 2016-01-07:
I've been working on this little client program and the library worked fine with the 0.9.3 python server but once I updated the server to the latest found in pip repositories this problem appeared. If I try to request any information about the Xml Schema node via getters / etc, all hell is loose.

Everything else seems to work fine though and if I just request the Objects node instead of root everything works fine and the new python server is even faster than the older version.

Below is a screenshot;
[image]

oroulet, 2016-01-07:
the easiest thing to do is probably to use wireshark.
clone python opcua, reset to 9.3 tag, record a working session with wireshark
back to 0.10.3, record a session and look at the packet that hang. Must be something different there.
do not forget to set the opcua filter in wireshark

Unfortunately, looks like nobody has been working on the C++ lib the last months... and I only use the python lib for the time being

Harha, 2016-01-08:
Well, I found the exact ReadRequest from the client to the server that causes the misbehaviour, after that the server responds with a huge message (Length: 1514) with weird unidentifiable data. Haven't figured out a solution yet and I doubt that I will since I'm not too experienced with networking stuff, but we'll see. I'm not even sure if it's the python server that is misbehaving because it seems the the request the client sends is fine, but the response it gets back is incorrect, though obviously it is more likely to be just the C++ library just not supporting everything.

Below are the captured packets between both versions (0.9.20 & 0.10.3), filtered to show only the opcua messages;
[opcua_packets_filtered.zip](https://github.com/FreeOpcUa/freeopcua/files/83018/opcua_packets_filtered.zip)

Also I tried to connect to a Beckhoff OPC UA server with this, and as it reads in the README it works, but some of the Nodes aren't read properly. So there's definitely a lot to improve with this library, it's still great though!

oroulet, 2016-01-08:
Looks like you are reading node i=8252. And yes this node contains a huge bytestring. I have no idea why but it is part of the spec..... In fact we to remove that node value from C++ server because it was giving problems on small memory machines...
I also see that a lot of fragments messages are sent, I am guessing this is due to the server sending fragments instead of one big message. The python implementation support fragments (added after 9.3 I guess) but not the C++ library yet..

@alkor maybe you know why the server is sending fragment? and how to configure it or the client to not sent fragment. I do not remember how that stuff works

Concerning beckoff, did you try C++ implementation or python? I know we had some issues with beckoff. I need to look at it. I have at least one beckoff  PLC at work...

> So there's definitely a lot to improve with this library, it's still great though!

opc-ua is hopelessly complex protocol, the C++ implementation has some rough edges, but the python stuff should be really close to complete now

Harha, 2016-01-08:
I tested connecting to the Beckhoff OPC UA server with the C++ client libraries. Also you don't need to own a Beckhoff PLC to test their products, you can just install Visual Studio (not express) and then download and install the Beckhoff Twincat 3 XAE and OPC UA server package and by that point the server should already be running on your PC. I'm doing testing like that inside a Windows virtual machine, Beckhoff is great in the sense that they make testing their products easy and free.

Thanks for your quick responses and all the info, I'll see what I can do myself and will hopefully send some pull requests to the respository if I get anything fixed / working properly, in the near future.

alkor, 2016-01-08:
Yes, the difference between these two captures is that 0.10 server split a huge message into chunks. The easiest way to make client work again is to increase buffer sizes in Hello message (say, to 500000 bytes). I'll try to port my patches to C++ when I'll have some time.

By the way, this huge value is a base-64 encoded file opc.ua.types.xsd (about 200 kbytes).