FreeOpcUa/freeopcua #187: Problems installing FreeOpcUa

JoHo3695, 2016-04-11:
Hey there!
I am a real newcomer in programming Raspberry Pis. Still, I have to set up OPC-UA communication with a Raspi as a student research project. More precisely, the Raspi will have to function as client and to read out data of a PLC via OPC-UA. I already tried all the proposed ways to properly install the library but I just cannot get it running.
Can anybody help me out with something like a step-by-step tutorial how to set it up correctly? I would really appreciate that!

Thank you very much in advance :)

destogl, 2016-04-12:
Some more data about the problem, would be nice. What are you trying to start, which PLC are you trying to talk to, which OPC UA server is installed on PLC?

JoHo3695, 2016-04-21:
I want to realize a OPC-UA client on a Raspberry Pi using the c++ library. More precisely, I want to read boolean sensor values out of the CPU of a Rexroth Indra Logic L 45 PLC. The installed server uses the binary Protocoll (opc.tcp://xx.xx.xx.xxx:port). Therefore, I am trying to set up the library correctly, but the installation routines fail. As a result, compiling the example files fails as well.

destogl, 2016-04-22:
I can not help you regarding failure without further informations about failure...

JoHo3695, 2016-04-22:
Using Ubuntu commands as it is described in debian.soft the failure is the following:

In file included from /usr/include/boost/thread/pthread/mutex.hpp:14:0,
                 from /usr/include/boost/thread/mutex.hpp:16,
                 from /usr/include/boost/thread/pthread/shared_mutex.hpp:12,
                 from /usr/include/boost/thread/shared_mutex.hpp:16,
                 from src/server/address_space_internal.h:22,
                 from src/server/address_space_internal.cpp:11:
/usr/include/boost/thread/xtime.hpp: At global scope:
/usr/include/boost/thread/xtime.hpp:88:1: error: expected declaration before '}' token
 } // namespace boost
 ^
Makefile:2321: recipe for target 'src/server/libopcuaserver_la-address_space_internal.lo' failed
make[2]: *** [src/server/libopcuaserver_la-address_space_internal.lo] Error 1
make[2]: Leaving directory '/usr/local/src/freeopcua'
Makefile:3441: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/usr/local/src/freeopcua'
Makefile:986: recipe for target 'all' failed
make: *** [all] Error 2

In case of using cmake installation fails like the following:

pi@raspberrypi /usr/local/src/freeopcua/build $ sudo cmake ..
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Found LibXml2: /usr/lib/arm-linux-gnueabihf/libxml2.so (found version "2.8.0") 
-- LibXML2 INCLUDE DIR IS: /usr/include/libxml2
-- Boost version: 1.49.0
-- Found the following Boost libraries:
--   system
--   program_options
--   filesystem
--   thread
-- Boost INCLUDE DIR IS: /usr/include
-- Boost LIBRARY DIR IS: /usr/lib
CMake Error at CMakeLists.txt:168 (target_compile_options):
  Unknown CMake command "target_compile_options".

-- Configuring incomplete, errors occurred!

Any ideas?

destogl, 2016-04-22:
Which version of boost and cmake do you have?

JoHo3695, 2016-04-22:
cmake 2.8.9
boost 1.49.0

GonzaloMMD, 2016-06-14:
Hi, JoHo,
you need a greater version of cmake ( >=2.8.12), in case you haven't solved it yet.