FreeOpcUa/freeopcua #188: First test on SL7: compiled, installed, crashed

DirkHoffmann, 2016-05-02:
I have installed the master head on `Scientific Linux release 7.1 (Nitrogen)`, `Linux myhost 3.10.0-229.1.2.el7.x86_64 #1 SMP Thu Mar 26 09:10:25 CDT 2015 x86_64 x86_64 x86_64 GNU/Linux`. Compilation went smoothly, and I started the `opcuaserver` which resulted of it.

I inspected the server namespace with [UaExpert](https://www.unified-automation.com/products/development-tools/uaexpert.html). (You can get its binaries for free, by simply registering to the provider Unified Automation.) After some successful browsing, the server crashed with its following last words:

```
02    3f(?) 00    00    03    3f(?) 00    00    04    3f(?) 00    00    05    3f(?) 00    00   
06    3f(?) 00    00    07    3f(?) 00    00    08    3f(?) 00    00    09    3f(?) 00    00   
0a    3f(?) 00    00    0b    3f(?) 00    00    0c    3f(?) 00    00    0d    3f(?) 00    00   
0e    3f(?) 00    00    0f    3f(?) 00    00    10    3f(?) 00    00    11    3f(?) 00    00   
12    3f(?) 00    00    13    3f(?) 00    00    14    3f(?) 00    00    15    3f(?) 00    00   
16    3f(?) 00    00    17    3f(?) 00    00    18    3f(?) 00    00    19    3f(?) 00    00   
1a    3f(?) 00    00    1b    3f(?) 00    00    1c    3f(?) 00    00    1d    3f(?) 00    00   
1e    3f(?) 00    00    1f    3f(?) 00    00    20    3f(?) 00    00    21(!) 3f(?) 00    00   
22(") 3f(?) 00    00    23(#) 3f(?) 00    00    24($) 3f(?) 00    00    25(%) 3f(?) 00    00   
26(&) 3f(?) 00    00    27(') 3f(?) 00    00    28(() 3f(?) 00    00    29()) 3f(?) 00    00   
2a(*) 3f(?) 00    00    2b(+) 3f(?) 00    00    2c(,) 3f(?) 00    00    2d(-) 3f(?) 00    00   
2e(.) 3f(?) 00    00    2f(/) 3f(?) 00    00    30(0) 3f(?) 00    00    31(1) 3f(?) 00    00   
32(2) 3f(?) 00    00    33(3) 3f(?) 00    00    34(4) 3f(?) 00    00    35(5) 3f(?) 00    00   
36(6) 3f(?) 00    00    37(7) 3f(?) 00    00    38(8) 3f(?) 00    00    39(9) 3f(?) 00    00   
3a(:) 3f(?) 00    00    3b(;) 3f(?) 00    00    3c(<) 3f(?) 00    00    3d(=) 3f(?) 00    00   
3e(>) 3f(?) 00    00    3f(?) 3f(?) 00    00    40(@) 3f(?) 00    00    41(A) 3f(?) 00    00   
42(B) 3f(?) 00    00    43(C) 3f(?) 00    00    44(D) 3f(?) 00    00    45(E) 3f(?) 00    00   
46(F) 3f(?) 00    00    47(G) 3f(?) 00    00    48(H) 3f(?) 00    00    49(I) 3f(?) 00    00   
4a(J) 3f(?) 00    00    4b(K) 3f(?) 00    00    4c(L) 3f(?) 00    00    4d(M) 3f(?) 00    00   
4e(N) 3f(?) 00    00    4f(O) 3f(?) 00    00    50(P) 3f(?) 00    00    51(Q) 3f(?) 00    00   
52(R) 3f(?) 00    00    53(S) 3f(?) 00    00    54(T) 3f(?) 00    00    55(U) 3f(?) 00    00   
56(V) 3f(?) 00    00    57(W) 3f(?) 00    00    58(X) 3f(?) 00    00    59(Y) 3f(?) 00    00   
5a(Z) 3f(?) 00    00    5b([) 3f(?) 00    00    5c(\) 3f(?) 00    00    5d(]) 3f(?) 00    00   
5e(^) 3f(?) 00    00    5f(_) 3f(?) 00    00    60(`) 3f(?) 00    00    61(a) 3f(?) 00    00   
62(b) 3f(?) 00    00    63(c) 3f(?) 00    00    64(d) 3f(?) 00    00    65(e) 3f(?) 00    00   
66(f) 3f(?) 00    00    67(g) 3f(?) 00    00    68(h) 3f(?) 00    00    69(i) 3f(?) 00    00   
6a(j) 3f(?) 00    00    6b(k) 3f(?) 00    00    6c(l) 3f(?) 00    00    6d(m) 3f(?) 00    00   
6e(n) 3f(?) 00    00    6f(o) 3f(?) 00    00    70(p) 3f(?) 00    00    71(q) 3f(?) 00    00   
72(r) 3f(?) 00    00    73(s) 3f(?) 00    00    74(t) 3f(?) 00    00    75(u) 3f(?) 00    00   
76(v) 3f(?) 00    00    77(w) 3f(?) 00    00    78(x) 3f(?) 00    00    79(y) 3f(?) 00    00   
7a(z) 3f(?) 00    00    7b({) 3f(?) 00    00    7c(|) 3f(?) 00    00    7d(}) 3f(?) 00    00   
7e(~) 3f(?) 00    00    7f() 3f(?) 00    00    80    3f(?) 00    00    81    3f(?) 00    00   
82    3f(?) 00    00    83    3f(?) 00    00    84    3f(?) 00    00    85    3f(?) 00    00   
86    3f(?) 00    00    87    3f(?) 00    00    88    3f(?) 00    00    89    3f(?) 00    00   
8a    3f(?) 00    00    8b    3f(?) 00    00    8c    3f(?) 00    00    8d    3f(?) 00    00   
8e    3f(?) 00    00    8f    3f(?) 00    00    90    3f(?) 00    00    91    3f(?) 00    00   
92    3f(?) 00    00    93    3f(?) 00    00    94    3f(?) 00    00    95    3f(?) 00    00   
96    3f(?) 00    00    97    3f(?) 00    00    98    3f(?) 00    00    99    3f(?) 00    00   
9a    3f(?) 00    00    9b    3f(?) 00    00    9c    3f(?) 00    00    9d    3f(?) 00    00   
9e    3f(?) 00    00    9f    3f(?) 00    00    a0    3f(?) 00    00    a1    3f(?) 00    00   
a2    3f(?) 00    00    a3    3f(?) 00    00    a4    3f(?) 00    00    a5    3f(?) 00    00   
a6    3f(?) 00    00    a7    3f(?) 00    00    a8    3f(?) 00    00    a9    3f(?) 00    00   
aa    3f(?) 00    00    ab    3f(?) 00    00    ac    3f(?) 00    00    ad    3f(?) 00    00   
ae    3f(?) 00    00    af    3f(?) 00    00    b0    3f(?) 00    00    b1    3f(?) 00    00   
b2    3f(?) 00    00    b3    3f(?) 00    00    b4    3f(?) 00    00    b5    3f(?) 00    00   
b6    3f(?) 00    00    b7    3f(?) 00    00    b8    3f(?) 00    00    b9    3f(?) 00    00   
ba    3f(?) 00    00    bb    3f(?) 00    00    bc    3f(?) 00    00    bd    3f(?) 00    00   
be    3f(?) 00    00    bf    3f(?) 00    00    c0    3f(?) 00    00    c1    3f(?) 00    00   
c2    3f(?) 00    00    c3    3f(?) 00    00    c4    3f(?) 00    00    c5    3f(?) 00    00   
c6    3f(?) 00    00    c7    3f(?) 00    00    c8    3f(?) 00    00    c9    3f(?) 00    00   
ca    3f(?) 00    00    cb    3f(?) 00    00    cc    3f(?) 00    00    cd    3f(?) 00    00   
ce    3f(?) 00    00    cf    3f(?) 00    00    d0    3f(?) 00    00    d1    3f(?) 00    00   
d2    3f(?) 00    00    d3    3f(?) 00    00    d4    3f(?) 00    00    d5    3f(?) 00    00   
d6    3f(?) 00    00    d7    3f(?) 00    00    d8    3f(?) 00    00    d9    3f(?) 00    00   
da    3f(?) 00    00    db    3f(?) 00    00    dc    3f(?) 00    00    dd    3f(?) 00    00   
de    3f(?) 00    00    df    3f(?) 00    00    e0    3f(?) 00    00    e1    3f(?) 00    00   
e2    3f(?) 00    00    e3    3f(?) 00    00    e4    3f(?) 00    00    e5    3f(?) 00    00   
e6    3f(?) 00    00    e7    3f(?) 00    00    e8    3f(?) 00    00    e9    3f(?) 00    00   
ea    3f(?) 00    00    eb    3f(?) 00    00    00    ec    3f(?) 00    00    84    5b([) a8   
1e    06    a3    d1    01    ff    ff    ff    ff    ff    ff    ff    ff    ff    ff    ff   
ff   
opc_tcp_async| Error during receiving message header: End of file
opc_tcp_async| Failed to send data to the client. Bad file descriptor
terminate called after throwing an instance of 'std::bad_weak_ptr'
  what():  bad_weak_ptr
Abort

```

Is there anything I can provide in addition for debug? Where should I start looking for reasons of the crash?

destogl, 2016-05-02:
It would be really helpful find out on which message is this happening. We are usually doing by sniffing the traffic with WireShark which has build in parser for OPC-UA messages. In that case you/we would be able to see which message is breaking the server. It is possible that this operations is not supported or nor fully/correctly implemented.

DirkHoffmann, 2016-05-11:
You may call it good news, but I was not able to reproduce the crash.

However there is a lot of debug output on the server, which is useless or should at least be optional. If I stop the output for some reason (e.g. by XOFF), then the communication with the client is dead for some time, and server crashes after it can go on again (XON). 
Not nice, not very stable, but probably not unexpected. Real servers should probably run in threads anyway.

DirkHoffmann, 2016-06-15:
Sorry to come back so lately after my last feedback. I hope this update is still useful for you.
After the crash, I restarted the server (and the UaExpert client) and let it run in background. Then I almost forgot it and had a look again on July 2nd. i.e. approximately after 3 weeks running. Both applications were still running fine.
So please close #188 for now.

closed 2016-06-15 by DirkHoffmann