FreeOpcUa/freeopcua #189: Files to be ignored

DirkHoffmann, 2016-05-02:
I have installed master head (43b86ae) and compiled it on SL7. There are a number of files being notified by git after compilation:

modified:   INSTALL
modified:   python/INSTALL
modified:   python/Makefile.in
modified:   python/missing
modified:   tests/gmock/Makefile.in
modified:   tests/gmock/build-aux/missing
modified:   tests/gmock/gtest/build-aux/missing
modified:   tests/gmock/gtest/m4/libtool.m4
modified:   tests/gtest/Makefile.in
modified:   tests/gtest/build-aux/missing
modified:   tests/gtest/m4/libtool.m4

Untracked files:
debian/changelog
example_client
example_server
install-sh
libopcuaclient.la
libopcuaclient.pc
libopcuacore.la
libopcuacore.pc
libopcuaprotocol.la
libopcuaprotocol.pc
libopcuaserver.la
libopcuaserver.pc
libtool
ltmain.sh
opcua
opcuaserver
stamp-h1
test-driver
tests/gmock/gtest/libtool
tests/gmock/gtest/samples/libsamples.la
tests/gmock/gtest/scripts/gtest-config
tests/gmock/libtool
tests/gmock/scripts/gmock-config
tests/gtest/libtool
tests/gtest/samples/libsamples.la
tests/gtest/scripts/gtest-config

Would be nice to have git ignore them (`.gitignore`?), in order to ease the work of people willing to contribute.

destogl, 2016-05-02:
Hi,

Thanks for your interests in freeopcua.

I am using the freeopcua inside other project and compile files are than created on other location so I never realised there are such files.

I think you are right, most of this files can be ignored. Could you create PR

DirkHoffmann, 2016-05-02:
Today (4:02am) Denis ?togl wrote:

> I think you are right, most of this files can be ignored. Could you create
> PR

OK, I will make a proposal. It will be easy for untracked files.

But there are files like `python/Makefile.in` and `tests/gtest/Makefile.in`, 
which are generated from corresponding `Makefile.am` files. Do the `.in` 
files have to be in git for some reason? Maybe there is some subtlety I 
missed?

oroulet, 2016-05-02:
We discussed this a year ago for generated code files and decided to keep the generated files under git since diff were interesting. I have no opinion about .in files, my opinion is that autotools should be removed and just keep cmake

DirkHoffmann, 2016-05-11:
> decided to keep the generated files under git since diff were interesting
But the git notification that they have changed is confusing, especially when compiling on systems without the latest versions of all packages. Hence it will be difficult to make clean pull requests, except for people who can afford to develop with exactly the latest environment that is used upstream.

> my opinion is that autotools should be removed and just keep cmake
I see only instructions for `autoreconfig && make`. But I agree that the situation is confusing to have several build configuration files in the project.

oroulet, 2016-05-11:
OK remove them