FreeOpcUa/freeopcua #191: How tu use 'enum class VariableAccessLevel'

bkuner, 2016-05-25:
to create the mask the definition of the enum class doesn't allow to OR the single values to the mask. I would suggest to add the following operator to the file variable_access_level.h:

inline uint8_t operator|(VariableAccessLevel x, VariableAccessLevel y)
{
    return (uint8_t) ((uint8_t)x | (uint8_t)y);
}

oroulet, 2016-05-26:
You are right. The operator is missing for many enum. These enums are auto generated from a python file under schema, if you want to try to fix it.

bkuner, 2016-05-26:
Sorry, I  never dealed with the python stuff here. If you could give me a hint where to put hands on, I could have a look on it.

bkuner, 2016-05-26:
I greped for 'enum class' and found out, that the enums that looks for to be used as mask are in enums.h. There the operator| and operator& is defined. The only missing one is the one in variable_access_level.h

oroulet, 2016-05-26:
A few enums are hand written. Looks like you found one 😃. Should be easy
to fix then :-)

bkuner, 2016-05-27:
Hi, 
yesterday I had a look on your enum.h and found out, that you define enums of type mask and others of type choice (and may be more types I didnt recognize), but you defined &, | operators for all. This operators are reasonable for masks but not for choices. Another point is that the operators use casts. Specially you cast the result of the operation to something that is not part of the enum. This is not the idea of it. You want to be sure that your masks and your choices use ONLY the values defined in the corresponding 'enum class' and you want type checking for the sort of int that represents the the mask.
 First I thougt ok, forgetit, but this  morning 4:30am I had the Idea to this solution - it is not easy to be a born nerd: 

Add the reasonable methods for mask and choice by templates: Type T is the enum class, Type U is the representation type.

```
template <class T, class U> class Mask  {
public:
    inline Mask set(T x) { mask |= (U) x; return *this;}
    inline Mask clr(T x) { mask &= ~((U) x); return *this;}
    inline bool isSet(T x)           { return (mask & ((U)x))?true:false; }
    inline U get(void)                   { return mask; }
private:
    uint32_t mask;
};

template <class T, class U> class Choice  {
public:
    inline Choice set(T x) { choice = (U) x; return *this;}
    inline bool   is(T x)           { return (choice == ((U)x))?true:false; }
    inline U get(void)                   { return choice; }
private:
    uint32_t choice;
};
```

Here is the Example how to be used.
```
enum class OpenFileMode : uint32_t
{
     Read = 1,
     Write = 2,
     EraseExisiting = 4,
     Append = 8
};

enum class ErrState : uint8_t
{
     Ok = 0,
     FeelBad = 1,
     Warn = 2,
     Error = 3,
     Desaster = 4
};
int main(int argcnt,char **argv)
{
    Mask<OpenFileMode,uint32_t> m;
    Choice<ErrState,uint8_t> c
    // The mask
    m.set(OpenFileMode::Write).set(OpenFileMode::Read);
    if(m.isSet(OpenFileMode::Write))
        std::cout<< "isWrite"<<std::endl ;
    if(m.isSet(OpenFileMode::Read))
        std::cout<< "isRead"<<std::endl ;
    m.clr(OpenFileMode::Read);
    if(! m.isSet(OpenFileMode::Read))
        std::cout<< "Read cleared mask: "<<m.get()<<std::endl ;
    //The choice
    c.set(ErrState::Ok);
    if(c.is(ErrState::Ok)) {
        std::cout<< "Choice OK: "<<c.get()<<std::endl ;
        c.set(ErrState::Warn);
    }
    if(! c.is(ErrState::Error)) {
        std::cout<< "Choice No Error but: "<<c.get()<<std::endl ;
        c.set(ErrState::Desaster);
    }
    std::cout<< "Choice Ends in: "<<c.get()<<std::endl ;
}
```
I didn't work with c++ in the last 10 years so I don't know if there is allready solution in the std or boost library for that. This is just a suggestion for a solution. And I don't see to have the time (besides that I'm afraid of:) to implement this in the whatever python script that creates enum.h and all places that use one of the enum classes.

oroulet, 2016-05-28:
I do not have time to look at this either. But we leave this open. Hopefully someone is interested. There is a bunch of cleanup to do in this lib