FreeOpcUa/freeopcua #194: Client/Server Certificates

iquestor, 2016-07-11:
I am trying to get the sample client on a Raspberry Pi to connect to a remote Kepware 5.19 OPCUA Server. I know I need to have certificates but don't know where they are, or are supposed to be on the client side; I am assuming I'll have a client side certificate in some place and then import this into the Kepware Server, and then add the client as a trusted client.  

I would appreciate it if someone could tell me where the certificates are/or should be on client side

fvacek, 2017-11-22:
The certificates authentication is not currently implemented in the freeopcua c++ binary client

fvacek, 2017-12-13:
The certificates authentication is currently implemented in the freeopcua c++ binary client. Server ends certificate to client to enable RSA-OAEP password encryption. see: 6ae743c21bd5ec8633141ff69d86fbecd2f67dce

shivakumar-tekumatla, 2025-07-23:
@fvacek I do not see this usage in the example file. Can you please update if you get a chance? Thanks in advance.