FreeOpcUa/freeopcua #198: Compiling on Fedora 24

caspervanzoest, 2016-09-09:
**When compiling the code on Fedora 24 - i386 architecture,
using GNU autotools:
following error occurs:**
`libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I./include -I/usr/include/libxml2 -O0 -g -std=c++11 -MT src/protocol/libopcuaprotocol_la-binary_stream.lo -MD -MP -MF src/protocol/.deps/libopcuaprotocol_la-binary_stream.Tpo -c src/protocol/binary_stream.cpp -o src/protocol/libopcuaprotocol_la-binary_stream.o >/dev/null 2>&1
mv -f src/protocol/.deps/libopcuaprotocol_la-binary_stream.Tpo src/protocol/.deps/libopcuaprotocol_la-binary_stream.Plo
make[2]: *** No rule to make target 'src/protocol/binary_attribute.cpp', needed by 'src/protocol/libopcuaprotocol_la-binary_attribute.lo'.  Stop.
make[2]: Leaving directory '/home/lray/Desktop/freeopcua-master'
Makefile:3558: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/lray/Desktop/freeopcua-master'
Makefile:1271: recipe for target 'all' failed
make: *** [all] Error 2
`

**as suggested, using CMake:**
`CMake Error at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find LibXml2 (missing: LIBXML2_LIBRARIES LIBXML2_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake/Modules/FindLibXml2.cmake:69 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:96 (find_package)
`

**Please note all required packages are installed (including libxml2).**

Any suggestion is welcome.
Thanks in advance!

mdcb, 2016-09-29:
You'll need the `libxml2-devel` package.

mdcb, 2016-10-11:
Please re-open if you need more help. Comments if you fixed it very much welcomed too.
Having gone through it, I cannot find any particular issue on f24.

closed 2016-10-11 by mdcb