FreeOpcUa/freeopcua #199: Server and Client in Same binary

girish99, 2016-09-13:
Hi,
     This is Girish , im new To c++ and i successfully connected a client to opcua server running on my Plc. And now I want to implement a server working along with the client on same binary. Is it possible?? If so any suggestions please. I wrote code on example-server and it shows error of undefined reference to opcua client.... And when I wrote server code in example-client the error is undefined reference to opcua server.

Regards
Girish

oroulet, 2016-09-13:
As far as I know, it should work...

destogl, 2016-09-13:
Hi,

you should have 2 instances, one for server and one for client. And probably different endpoints

girish99, 2016-09-20:
hi guys, than you.Thats working good. need 2 different endpoints, no needof 2 instances. Have o mae changes in cmakefiles like adding required libraries.

destogl, 2016-09-27:
We are glad it is solved!

closed 2016-09-27 by destogl