FreeOpcUa/freeopcua #201: Aloate Memmory Error

SimarglFox, 2016-09-22:
Debian amd64
latest version of Freeopcua
cross compiller for armhf (gcc 4.9.2)
cmake version 3.0.2
6gb RAM

Then i try to build project i have error alocate memory here is log:

[ 76%] Building CXX object CMakeFiles/opcuaserver.dir/src/server/services_registry_factory.cpp.o
[ 77%] Building CXX object CMakeFiles/opcuaserver.dir/src/server/services_registry_impl.cpp.o
[ 78%] Building CXX object CMakeFiles/opcuaserver.dir/src/server/standard_address_space_part3.cpp.o
[ 80%] Building CXX object CMakeFiles/opcuaserver.dir/src/server/standard_address_space_part4.cpp.o
[ 81%] Building CXX object CMakeFiles/opcuaserver.dir/src/server/standard_address_space_part5.cpp.o

cc1plus: out of memory allocating 4064 bytes after a total of 3633590272 bytes

oroulet, 2016-09-22:
yes this file is autogenerate and big... but people manage to compile it n
raspberry with the right options

SimarglFox, 2016-09-22:
Can you say this configurating?