FreeOpcUa/freeopcua #220: Issue with GetBrowseName().Name

ashubhatt, 2017-01-06:
expecting 9  received: 3
Unable to deserialize NodeId. Unknown encoding type received.
boost::bad_any_cast: failed conversion using boost::any_cast

oroulet, 2017-01-06:
you will need to find out which node is failing

ashubhatt, 2017-01-06:
Debug Enabled and got this:

```
expecting 9  received: 3
binary_client| Browse <--
binary_client| Read -->
srv=0;ns=1;s=13468a63-8791-4077-8c65-67c12f11ffd7/0:SessionDiagnostics;  3
binary_client| ReceiveThread : Error receiving data: Unable to deserialize NodeId. Unknown encoding type received.
```
And exception thrown: 
`Could not retrieve browse name.`

ashubhatt, 2017-01-07:
I am not sure where should I handle the error..  I am calling the **findNode** function to Browse through nodes of Prosys Server:
```
void findNode(OpcUa::Node &myNode,std::string c)
{
	for (OpcUa::Node node: myNode.GetChildren())
	{
		std::cout << c << node.GetBrowseName().Name << std::endl;
		findNode(node,c+"  >");
	}
}
```

The debug statements are as below:

```
binary_client| Browse -->srv=0;ns=2;s=MyLevel.Alarm;  
binary_client| Got response id: ns=0;i=530; and handle 441
expecting 4  received: 3
binary_client| Browse <--
binary_client| Read -->
srv=0;ns=2;s=MyMethod;  3
expecting 2003782656  received: 177
binary_client| ReceiveThread : Error receiving data: Not enough data was received from channel.

Exception:: Could not retrieve browse name.
```
Can someone point me in right direction?

oroulet, 2017-01-07:
Now you are writing about another node. To debug that kind of things wireshark is a good tool. Can you try with it? It has support for opcua. thus we can see every bytes going through and find out what we cannot parse

ashubhatt, 2017-01-09:
I have tried analyzing the packets with wireshark. I think the opc client stops functioning when it encounters fragmented message during **_BrowseRequest_** as can be seen from the screenshot.

[image]

**The console output with debug enabled is attached below:**

[image]

Further in the Wireshark screenshot, one can see that ReadRequest is fired even before the BrowseResponse is available. Shouldn't it ideally wait for BrowseResponse first and then fire ReadRequest? Please let me know if you need any more details on this topic. I am really stuck in this portion.

oroulet, 2017-01-09:
great debugging! Unfortunately fragments are not implemented in C++. Patch
are welcome if you get some time.

It is implemented in the python version if you want something to start from
https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/ua/uaprotocol_hand.py
The code is quite hard to read- Ideas to clean this uo are also welcome

ashubhatt, 2017-01-09:
Ok.. Can you point me as in which file and what function do I need to modify to handle fragmented messages?

oroulet, 2017-01-09:
https://github.com/FreeOpcUa/freeopcua/blob/master/src/client/binary_client.cpp

Look at the receive method. you need to add another case and buffer data if
it is fragmented.

ashubhatt, 2017-01-10:
I have modified the Receive function as you suggested. Also modified the BufferSize because we are getting fragments with length more than 4096.
Sent a PR, kindly let me know if any changes are required.

closed 2017-01-16 by ashubhatt