FreeOpcUa/freeopcua #221: Issue with disconnection alert

ashubhatt, 2017-01-06:
Is there a way to get an alert in the client code when connection to server is broken? 

I would like to perform a reconnection upon receiving a disconnect alert. If possible kindly provide a pseudo code (or snippet) for the same.

oroulet, 2017-01-06:
This has been discussed a few times. There is no way to get a push notfication from server so the most reliable solution is to poll. This is what all other clients I have seen do.
In you main thread, read a node value, (for example server state node), and reconnect if there is any error.

ashubhatt, 2017-01-06:
Hey @oroulet,

I have modified binary_client.cpp & client.cpp slightly such that BinaryClient's constructor accepts a std::function called disconnectCB.
When an exception is raised because of disconnection, I simply call this disconnectCB function which is available in client.cpp where user can write his custom code.

In disconnectCB I've tried to reconnect with the server at an interval of 5 seconds. This solves the problem without polling the node value.
If you require, I can send a PR.

closed 2017-01-16 by ashubhatt