FreeOpcUa/freeopcua #223: Issue with client code for proper disconnection if server crashes

nabilgandhi, 2017-01-16:
The client code hangs if the server gets disconnected or crashes.Is there a proper way to disconnect client if the server crashes?
I have tried polling but that consumes my thread.so I am looking for more optimum solution.

destogl, 2017-02-22:
As far as I know only possibility is polling or subscribing on some server variable (in the background still polling)

oroulet, 2017-02-22:
Yes polling is the only possibility. This is a socket limitation I guess

closed 2017-02-22 by oroulet

0-Avalon-0, 2022-09-13:
Sorry for interrupting again, I have a similar problem about reconnection. I did subscription to an opcua server which is on a PLC in my connection method. I also wrote a reconnection thread to invoke connection method again in case the opcua server is disconnected or crashed. The reconnection thread works, but there is a memory problem as every time I shutdown opcua server and restart it, some memory is allocated without being released. It seems that memory is allocated in the subscription process, but I don't know how to release the memory when server is disconnected. 

I know this might be stupid, just hope someone could help me with this. Thanks a lot.

guojing555, 2022-09-13:
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。