FreeOpcUa/freeopcua #228: opc/ua/protocol/types_manual.h isn't installed

roelschroeven, 2017-03-08:
When trying to build example_client.cpp:

    $ make
    Scanning dependencies of target test1
    [100%] Building CXX object CMakeFiles/test1.dir/example_client.cpp.o
    In file included from /usr/local/include/opc/ua/protocol/protocol.h:3:0,
                     from /usr/local/include/opc/ua/protocol/session.h:15,
                     from /usr/local/include/opc/ua/services/services.h:14,
                     from /usr/local/include/opc/ua/node.h:22,
                     from /usr/local/include/opc/ua/client/client.h:23,
                     from /home/roel/src/testjes/freeupcua-test1/example_client.cpp:11:
    /usr/local/include/opc/ua/protocol/protocol_auto.h:20:42: fatal error: opc/ua/protocol/types_manual.h: No such file or directory
     #include <opc/ua/protocol/types_manual.h>

It seems `make install` doesn't install that file.

mdcb, 2017-03-09:
Interesting. In #227 you are using autotools. Are you using CMake here?
PR #226 (in-progress) should fix the issue for autotools.

mdcb, 2017-03-09:
PR #226 has been merged. Please try again if you are using autotools.

mdcb, 2017-03-17:
Reopen if #226 did not fix your issue, closing for now.

closed 2017-03-17 by mdcb