FreeOpcUa/freeopcua #233: install error

aquari89, 2017-03-10:
Hello, All.
I have a problem.
I want to install FreeOpcUa(C++) using cmake.
''`mkdir build
cd build
cmake ..
make
make test'''

I followed the instructions, but an error occurred.
I tried several times, but it is the same.
Do I have something wrong?

'''
test@test-virtual-machine:~/freeopcua-master/build$ cmake ..
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
CMake Error at /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:108 (message):
  Could NOT find LibXml2 (missing: LIBXML2_LIBRARIES LIBXML2_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:315 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-2.8/Modules/FindLibXml2.cmake:62 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:100 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/keti/freeopcua-master/build/CMakeFiles/CMakeOutput.log".
'''

mdcb, 2017-03-10:
You need to install the libxml2 development package. Different OS do it differently e.g., redhat: ``sudo yum install libxml2-devel``, or more recently on fedora: ``sudo dnf install libxml2-devel``.
Look it up for your OS/distribution.

aquari89, 2017-03-10:
@mdcb 
Thank you! I solved it!

closed 2017-03-10 by mdcb

WangAllen, 2018-04-16:
CMake Error at /usr/share/cmake-3.5/Modules/FindBoost.cmake:1677 (message):
  Unable to find the requested Boost libraries.

  Boost version: 1.58.0

  Boost include path: /usr/include

  Could not find the following Boost libraries:

          boost_system
          boost_filesystem
          boost_thread

  Some (but not all) of the required Boost libraries were found.  You may
  need to install these additional Boost libraries.  Alternatively, set
  BOOST_LIBRARYDIR to the directory containing Boost libraries or BOOST_ROOT
  to the location of Boost.
Call Stack (most recent call first):
  CMakeLists.txt:108 (find_package)

-- Boost INCLUDE DIR IS: /usr/include
-- Boost LIBRARY DIR IS: /usr/lib/x86_64-linux-gnu
-- SSL support using libmbedtls: ON
--   mbedtls LIBRARIES: mbedx509mbedtlsmbedcrypto
-- Compiling module for python 3.5.2
-- Could NOT find Boost
-- Boost python3 lib not found: Not building python3 module
-- Configuring incomplete, errors occurred!
See also "/home/python/Desktop/freeopcua-master/build/CMakeFiles/CMakeOutput.log".
See also "/home/python/Desktop/freeopcua-master/build/CMakeFiles/CMakeError.log".

i have already install Boost package