FreeOpcUa/freeopcua #255: Error when trying to add a method node from a client

bassofil, 2017-06-22:
I'm trying to add a method **GetStatistics** calling `OpcUa::Node::AddMethod()` method from a client process, but the client keeps crashing.
Let's say that `OpcUa::Node::AddNode()` and `OpcUa::Node::AddVariable()` are working perfectly as expected, and that adding the same method directly from the server (no TCP data exchanged) is working well too.
Debugging the code a little bit, to me, it seems that the expected (by both client and server) message size is wrong, that is, the server actually tries to read more bytes than the ones available.

Client console output:
```
binary_client| ReceiveThread : Error receiving data: Failed to receive data from host. Connection reset by peer.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Response timed out
Aborted (core dumped)
```

Server console output:
```
opc_tcp_async| Received message header with size 8
opc_tcp_async| Message type: 6
opc_tcp_async| Chunk type: 1
opc_tcp_async| MessageSize: 169
opc_tcp_async| Waiting 161 bytes from client.
opc_tcp_async| Received 161 bytes from client:
Data size: 161
01    00    00    00    02    00    00    00    08    00    00    00    08    00    00    00   
01    00    e8    01    02    00    00    07    00    00    00    a4    9e    c7    b1    2e(.)
eb    d2    01    08    00    00    00    00    00    00    00    ff    ff    ff    ff    10   
27(') 00    00    00    00    00    01    00    00    00    02    00    00    55(U) 00    00   
00    02    00    00    2f(/) 00    00    00    02    02    00    23(#) 4e(N) 00    00    02   
00    0d    00    00    00    47(G) 65(e) 74(t) 53(S) 74(t) 61(a) 74(t) 69(i) 73(s) 74(t) 69(i)
63(c) 73(s) 04    00    00    00    01    00    68(h) 01    01    30(0) 00    00    00    60(`)
01    16    00    02    0d    00    00    00    47(G) 65(e) 74(t) 53(S) 74(t) 61(a) 74(t) 69(i)
73(s) 74(t) 69(i) 63(c) 73(s) 02    0d    00    00    00    47(G) 65(e) 74(t) 53(S) 74(t) 61(a)
74(t) 69(i) 73(s) 74(t) 69(i) 63(c) 73(s) 00    00    00    00    00    00    00    00    01   
01   
opc_tcp_processor| Processing secure message.
opc_tcp_processor| Processing 'Add Nodes' request.
expecting 2  received: 0
opc_tcp_async| Failed to process message. Not enough data was received from channel.
```

destogl, 2017-06-22:
I am not sure any more is methods are supported at all in this stack.

@mdcb do you know this?

oroulet, 2017-06-22:
I do not remember either...

mdcb, 2017-06-22:
@destogl I'm not familiar with this. Sorry, can't help.