FreeOpcUa/freeopcua #256: How to cross-build c++ freeOpcUa in Ubuntu to use for embedded ARM?

QHWG67, 2017-07-13:
We would like to build this c++ freeOpcUa in Ubuntu OS and use the output to run in embedded ARM chip, how should I do? Could you pls give some suggestions?

RavilN, 2017-07-13:
One of the options is VisualGDB: https://visualgdb.com/
With it, you can use Visual Studio (community edition) as main IDE, and
from there use Ubuntu VM to build binaries for ARM target, then run and
debug on your embedded device

QHWG67, 2017-07-13:
I would like to use the arm-linux-gcc complier to build the whole c++ project in Ubuntu OS in VirtualBox,  use the results(.so etc) in embedded device. So how should I update the CMakeLists.txt file to use CMake to build them instead of using IDE? Many thanks!

DavidZemon, 2018-01-16:
This is a CMake question and not a freeopcua question. I recommend reading about cross-compiling with CMake and how to use Toolchain files. If you're a Docker fan, this can help too: https://github.com/dockcross/dockcross