FreeOpcUa/freeopcua #257: GetChild fails

ingarpedersen, 2017-07-28:
I'm having problems getting the getchild with a path to work.
I'm running a OPC foundation reference server, and a UaExpert client that can read the server just fine.

In my code I have:
```
     std::vector<std::string> varpath{ "Objects", "Server"}; 
      myvar = root.GetChild(varpath);
      std::cout << "got node: " << myvar << std::endl;
      for (OpcUa::Node node : myvar.GetChildren())
        { std::cout << "    " << node << " : " << node.GetBrowseName().Name << std::endl; }
```
Which amongst other things return:
    Node(ns=0;i=2255;) : NamespaceArray
    Node(ns=0;i=2256;) : ServerStatus
    Node(ns=0;i=2267;) : ServiceLevel

But if I change the varpath to:
`std::vector<std::string> varpath{ "Objects", "Server", "ServerStatus"}; `
I get a "BadNoMatch" status code

Same result if I try to use QualifiedName vector.

destogl, 2017-07-31:
Did you try to put NamespaceID before the string name, e.g. "2:Objects"

ingarpedersen, 2017-07-31:
Yes, 

`std::vector<std::string> varpath{ "0:Objects", "0:Server", "0:ServerStatus"};`

gives the same result... But 

`std::vector<std::string> varpath{ "0:Objects", "0:Server"};`

works just fine... And as you can see in the original post, there is a child in ns=0 with name=ServerStatus so there is something strange going on.