FreeOpcUa/freeopcua #267: Cannot start multiple Clients on one System, which seems to be cause by the new spdlog

RP-01, 2017-09-17:
Hello,

in a current project, I'm starting multiple instances of `OpcUa::UaClient`.
This worked fine until one of the most recent commits, which implemented `spdlog` into the FreeOpcUa implementation.
Every time a second UaClient is attempt to be started after the first one, this exception is thrown:
> exception: logger with name 'UaClient' already exists

I looked into the problem, and I think it's caused by this line in `freeopcua/src/client/client.cpp, UaClient:UaClient(bool debug)`:
`109  Logger = spdlog::stderr_color_mt("UaClient");`

`spdlog `will throw an exception in `spdlog/details/registry.h`, as soon as a new Logger with the same name is being registered.
The problem with this is, that there doesn't seem to be an option implemented in `spdlog::stderr_color_mt`, which just ignores this case, and uses the one already existing Logger available. Either we need to use one of the other Logger-Types provided, or this needs to be handled by the UaClient.

Can anyone reporduce this problem and confirm what I've noticed?

ralf1070, 2017-09-18:
You are right. I did it the way you propose in PR #268.

closed 2017-09-18 by ralf1070

RP-01, 2017-09-18:
Thank you very much!