FreeOpcUa/freeopcua #272: Checking for invalid nodes an a Server

RP-01, 2017-09-20:
Hello,

a short explanation of the problematic:
- If we want to subscribe to a DataChange event of a node, the node has to be existent on the serverside, because otherwise the client connected will throw an exception and exit.
- Using `client->GetNode( nodeID )` will always return a "valid" Node with the Namespace and ID given via the passed NodeID-Object, even if it doesn't exist on the serverside. (So if you have passed a NodeID with (ns=42;i=1337) you'll get a node with the exact same infromation back.
- If you now want to check for the validity of the node so the client doesn't crash while trying to subscribe, this is not possible in the -in my opinion- most intuitive way.

My solution for this is/was as follows:

1. `OpcUa::Node node = client->GetNode( nodeId ); // Get specific Node from Client`
2. `OpcUa::Node parentNode = node.GetParent(); // Get parent of this specific Node.`
3. `if ( !( parentNode.GetId().GetNamespaceIndex() == 0 && parentNode.GetId().GetIntegerIdentifier() == 0 ) ) // If the (Namespace and ID of the parent) don't equal zero -> node exists`
4. `{ //do stuff, subscribe stuff } else { throw std::exception( "Node doesn't exist." ); }`

This works, because `GetParent()` will return a `Node(0,0)`, if no parent could be found for the specific node. Since the specification of OPCUA didn't intend to have a `Node(0,0)` in the model, this should be relatively failsafe, too. But I think this is a little bit cumbersome, and could be improved.

So, how about we make the `GetChild( nodeID )` function work like the `GetParent()` function, and return a `Node(0,0)` whether the node could be found on the Server or not?
Does this interfere with any other part of the specification, or would this solution be acceptable, to make checking for the existence of a node a bit more convenient?

oroulet, 2017-09-20:
Sometimes nodes do not have parents and exist. The original way to test for
a nice existence was to call and method like GetBrowseName() then an
exception is raised if the node does not exist.

ralf1070, 2017-09-20:
As far as I understand the GetParent() thing is his workaround. The GetBrowseName() thing is - for what I think about it - a workaround too.

What about an IsValid() method on Node to have something less clumsy to check a Node?

oroulet, 2017-09-20:
What would the is valid method do? We need to call server anyway. So
GetBrowseName is not more expensive

ralf1070, 2017-09-20:
Actually IsValid() could call GetBrowseName() to do it's job. But IsValid() would be a much more obvious method to call for a user of this API and the resulting code would be much better to read:

```
if (!node.IsValid()) {
    cerr << "Houston, we've got a problem" << endl;
    return -1;
}

```
vs:

```
try {
    // GetBrowseName() throws an exception if node does not exist
    node.GetBrowseName()
} catch (Exception e) {
    cerr << "Houston, we've got a problem" << endl;
    return -1;
}
```

RP-01, 2017-09-20:
I think this would be a suitable solution.
The only thing left to verify then is whether the behavior for GetBrowseName() is defined on machines which are not completely browseable, too.
I know that Siemens for instance is a bit sloppy with their Node-Tree Models (maybe intentional), and some nodes aren't browseable by any name at all, but only reachable by their respective NodeIDs. So maybe we could run into conflicts there, if the returned Value doesn't equal `VariantType::QUALIFIED_NAME` (That's what GetBrowseName() checks for).

Maybe we could even use `OpcUa::BinaryClient::Read(const ReadParameters & params)` directly for this to work, since it will return a std::vector of all Results found (if my understanding is right), like this:
```
bool Node::IsValid() const
{
    ReadParameters params;
    ReadValueId attribute;
    attribute.NodeId = Id;
    attribute.AttributeId = AttributeId::NodeId;
    params.AttributesToRead.push_back( attribute );

    std::vector<DataValue> vec = Server->Attributes()->Read (params );

    if ( vec.size() > 0 )
    {
        return true;
    }
    else
    {
        return false;
    }
}
```

This is just the idea, I'll check this out later as soon as I'm at home.

oroulet, 2017-09-20:
Maybe it is possible to implement Is valid by reading nodeid from server.
At least that one should be available!

RP-01, 2017-09-22:
I made a quick test with the solution above, but it didn't work reliable. (Nor with AttributeId::NodeID, neither with AttributeId::BrowseName)
I'll invest this further in the next few days, and also seek some inspiration from the original specification regarding the handling of non-existent nodes.

oroulet, 2017-09-22:
what is the issue? if the server does not let you query for NodeID or BrowseName it is completely broken

RP-01, 2017-09-22:
I never get any result returned. The vector was always empty.
I used a very basic and simple FreeOpcUa Server with just a few Nodes created.

oroulet, 2017-09-22:
Do you use the server or client from freeopcua?

RP-01, 2017-09-22:
Currently both, ans also with different FreeOpcUa Servers. But i've also evaluated it against a node- Server. The results were the same.

ralf1070, 2017-10-05:
Have a look at PR #278. Does this work for you?

harrypotter48, 2017-10-05:
thx, pero eso me dice si el nodo es valido, yo necesito que si la writeMask esta en 0 no se escriba en la propiedad y lance StatusCode::BadNotWritable
 

Enviar: jueves 5 de octubre de 2017 a las 10:33

ralf1070, 2017-10-05:
Sorry to say that - I don't understand what you write. What would you say when I start writing russian, or japanese or hindu. Write english or keep quiet.

Add on: using an online translator (deepl.com) I found, that you try to take over a thread to discuss a feature that is completely unrelated to this thread. Stop that immediately or I will drop all of your future posts without reading.

RP-01, 2017-10-05:
@ralf1070 
Looks great so far! I've tested it agaist multiple OPCUA Servers, and it seems to work without any problems.

Thank you very much, good work. In my opinion this Issue is solved.

closed 2017-10-05 by RP-01