FreeOpcUa/freeopcua #285: Add AlarmCondition fields to CustomEvent

awcullen, 2017-11-08:
What would it take to add the AlarmCondition fields to the Event type (or Custom EventType).

Could I add the properties to this call:
#Not working
alarmType = server.create_custom_event_type(idx, 'MyFirstAlarm', ua.ObjectIds.AlarmConditionType, [('0:ConditionId', ua.VariantType.NodeId), ('0:BranchId', ua.VariantType.NodeId), ('0:Retain', ua.VariantType.Boolean)])
  

I'm looking to have a server trigger alarm conditions that will show in UaExpert.

awcullen, 2017-11-09:
Found solution at https://github.com/FreeOpcUa/python-opcua/issues/278

ayonaphilipose, 2019-04-30:
Is it possible to implement client receiving alarms using python opcua?

awcullen, 2019-04-30:
First create a notification handler class:
```
class Handler(object):

    def datachange_notification(self, node, val, data):
        print("Data change: ", node, val)
    
    def event_notification(self, event):
        print("Event: ", event)

```
In your main:
```
        # Create subscription to server Events.
        sub = client.create_subscription(250, Handler())
       # Subscribe to receive BaseEventType from Server node.
        handle1 = sub.subscribe_events() 
```