FreeOpcUa/freeopcua #287: build freeopc

nrainbow, 2017-11-17:
i try to build freeopc for use with qtopcua. i want use it in a qt Project with qt 5.7 mingw530. i have install boost V 1.56.0
bootstrap.bat gcc
b2 link=shared toolset=gcc
if i try in cmd
"C:/Program Files (x86)/CMake/bin/cmake" .. -DBOOST_ROOT=C:/boost_1_56_0 -DBOOST_DEBUG=ON
then i have the following Output
Could not find the following Boost libraries:
          boost_system
          boost_program_options
          boost_filesystem
          boost_thread
Please can anyone help me

KeenRunner, 2018-01-13:
Have you solved the problem?
I have the same issue.

nrainbow, 2018-01-13:
No i couldnt solve it and i think the dependency of the solution from other librarys is the Problem. When i have a QT Project with mingw than is it a Problem to link librarys compiled with mscv. I would happy if anyone can  tell how can use this freopua solution with QT 5.x mingw.

Jokymon, 2018-01-27:
While trying to do almost the same thing myself I too ran into several issues. I never encountered the issue @nrainbow is describing but several other ones. So I decided to write up what I did here: https://hiseyesuncovered.wordpress.com/2018/01/28/building-freeopcua-with-mingw/ Maybe this helps anyone.

WangAllen, 2018-04-17:
i have the same issue in ubantu
Have you solved the problem?

cenit, 2019-03-22:
using [vcpkg](https://github.com/Microsoft/vcpkg) it should be possible to install freeopcua on windows and also mac/linux starting from sources, together with the most up-to-date boost libs. Please let me know if it works for you or any problem you may have, together we can fix it if there's any problem.