FreeOpcUa/freeopcua #299: keep getting Response timed out even thought it looks like I got it

codersmurf, 2018-03-10:
I added an extra debug message to the exception handler, but looking back at the logs I see the request sent and the reponse received.  I'm not sure if my code is exposing a bug, I hit the server pretty hard and getting ready to hit it harder but this came back.  Before I was getting before and handled it with waits before each getBrowseName from a name in my loops, it would still happen but I have lots of recovery stuff that kept it pretty stable, but after upgrading the monitoring server I can't get it stable no matter how much I wait.  Figured it was time to actually fix the issue.  I'm not sure if the locks are not matching up or the parse doesn't see it as the response even though the logs say it is.  Any help would be greatly appreciated. Thanks

codersmurf, 2018-03-10:
think I have a fix for the issue, logs look very promising, will cleanup, did you guys want the fix pushed up if it works out over the weekend?

oroulet, 2018-03-11:
We are always interested in fixes ;-)

codersmurf, 2018-03-11:
K, on Monday I can do this.  Never pushed to a git hub repository, what's
the procedure?  Also, the fix applies to the server code but I have no way
of testing this if I add it, so that stuff would be by Zen if u want it.
Thanks

codersmurf, 2018-03-13:
So what's the scoop on this, you guys want it? Thanks

codersmurf, 2018-09-08:
Never heard back and got busy and realized this was never resolved.  So I never did get push perms and figured I'd just attach the file here.  Let me know if you have any questions or concerns. Thx

[binary_client.cpp.zip](https://github.com/FreeOpcUa/freeopcua/files/2363009/binary_client.cpp.zip)

sschakravarty, 2019-02-21:
Is this issue resolved for you. Even I had similar bug when I used subscription of multiple tags and extracting data from OPCUA server. Please help!!

sschakravarty, 2019-02-21:
> Never heard back and got busy and realized this was never resolved. So I never did get push perms and figured I'd just attach the file here. Let me know if you have any questions or concerns. Thx
> 
> [binary_client.cpp.zip](https://github.com/FreeOpcUa/freeopcua/files/2363009/binary_client.cpp.zip)

Is this issue resolved for you. Even I had similar bug when I used subscription of multiple tags and extracting data from OPCUA server. Please help!!

codersmurf, 2019-02-21:
Well, if u use the code I posted it fixes the issue, it was never put into the main branch.  Try that, hopefully they push the code changes I made onto the main branch so it's resolved.

jason416, 2019-05-23:
@codersmurf thanks bro, I also get this problem. I added more logs to check out whether the Server send response packet, and it looks like the Server already sent for Client. the logs are:
> [2019-05-23 14:20:55.783] [UaClient] [info] binary_client         | wait for response, max time: 20000
[2019-05-23 14:20:55.786] [UaClient] [info] binary_client         | call response callback for message id: ns=0;i=634;, handle: 1045
[2019-05-23 14:20:55.786] [UaClient] [info] binary_client         | call OnData(), ready to wake up Send thread!
[2019-05-23 14:21:15.787] [UaClient] [info] binary_client         | timeout = 20, ready to throw Response timed out exception!
WARNING:root:Read error [opcua] @192.168.2.50 |address: ns=6;s=DataItem_0100 |code :Response timed out
WARNING:root:read error@192.168.2.50:Response timed out
[2019-05-23 14:21:15.794] [UaClient] [info] binary_client         | wait for response, max time: 20000
[2019-05-23 14:21:15.797] [UaClient] [info] binary_client         | call response callback for message id: ns=0;i=634;, handle: 1046
[2019-05-23 14:21:15.797] [UaClient] [info] binary_client         | call OnData(), ready to wake up Send thread!
[2019-05-23 14:21:15.798] [UaClient] [info] binary_client         | wakeup by OnData()
[2019-05-23 14:21:15.800] [UaClient] [info] binary_client         | wait for response, max time: 20000
[2019-05-23 14:21:15.803] [UaClient] [info] binary_client         | call response callback for message id: ns=0;i=634;, handle: 1047
[2019-05-23 14:21:15.803] [UaClient] [info] binary_client         | call OnData(), ready to wake up Send thread!
[2019-05-23 14:21:15.804] [UaClient] [info] binary_client         | wakeup by OnData()
[2019-05-23 14:21:15.806] [UaClient] [info] binary_client         | wait for response, max time: 20000

the problem is: In **OnData()**, response callback lambda function called by **Receive()**, it call **doneEvent.notify_all()**, sometimes the **Send()** was always can not be woke up by this.

what changes you did is more reliable to keep sender and receiver keep sync, in the case of no multi-thread call functions(which finally call Send()). It could be a nice patch :-)

codersmurf, 2019-05-24:
Cool, can u patch it for me, it's been so long since I looked at it might
take a few to get back in the grove on the source base:).

jason416, 2019-05-24:
> Cool, can u patch it for me, it's been so long since I looked at it might take a few to get back in the grove on the source base:).
> […](#)

Can you offer me a detail about what changes you did? And why ?