FreeOpcUa/freeopcua #300: Installing library ubuntu

uprightcarrion, 2018-03-12:
I'm having some trouble installing the library.  I'm using ubuntu so I ran the .soft script, I built it using autoconf, and ran make install.  Everything seems to have gone ok.   I copied the example server code but I can't get it to build.  

Problem 1: it couldn't find the headers so I manually moved the opc folder from the include folder to /usr/include.  The compiler was then able to find the headers.

Problem 2: Now that the headers are found I get build errors complain undefined references related to opcua.

I'm sure I'm doing it wrong.  I'm using CodeBlocks and haven't done anything special compiler configurations.

[buildErrors.txt](https://github.com/FreeOpcUa/freeopcua/files/1801114/buildErrors.txt)

f0urex, 2018-05-11:
I have this problem too

guqinchen, 2018-10-30:
@uprightcarrion I don't have problem 1 by calling
`sudo make install`
However, I do have problem 2. Did you eventually solve the problem?

uprightcarrion, 2018-11-01:
@guqinchen unfortunately no I wan't able to get past this.  I chopped it up to my lack of C/C++ experience.  I've been using the python wrapper.  I'd like to come back to this and try again, if you happen to get it figured out before I do post it here.  I'll do the same.

uprightcarrion, 2018-11-01:
@guqinchen Ok..some progress.  I was able to build the example_server.cpp with "g++ example_server.cpp -Llib -lopcuacore -lopcuaserver -lopcuaprotocol -lopcuaclient -lpthread" but the when I run the executable I get "./a.out: error while loading shared libraries: libopcuacore.so: cannot open shared object file: No such file or directory"

guqinchen, 2018-11-01:
@uprightcarrion I really appreciate your help! I can successfully build and run the example_server.cpp now.

For me, I observed the installation process first by following the "Using cmake - Linux" instruction in README, and found that I needed to install boost and mbedtls library. So I installed boost with
`sudo apt-get install libboost-all-dev`
and installed mbedtls with
'''
git clone https://github.com/ARMmbed/mbedtls.git
cd mbedtls
sudo make install
'''
Then freeopcua will install successfully with
'sudo make install'

I was initially stuck with the linking process during compiling the example server program. I was suspecting that gcc was able to find the header files but could not link to the library files, but was unsure how to solve this. Fortunately, your method worked for me! To narrow it down to the minimal argument set, I called
'g++ example_server.cpp -lopcuacore -lopcuaprotocol -lopcuaserver -lpthread -o example_server'
without further calling 'sudo ldconfig'. Now it seems to work just fine!

Also in case you are interested in an alternative (since this repo hasn't been updated for a while, unlike the Python version), I have already managed to use [Python embedding](https://docs.python.org/2/extending/index.html) to call the Python APIs and integrate with a C++ project. I know this sounds silly, but it was the only thing I could do and it worked.

uprightcarrion, 2018-11-01:
@guqinchen no problem, I'm glad I got it to work this time.  I guess I've learned a little more since last time I took a whack at it.  I'm sure there are probably better methods but I'll learn more as a go.