FreeOpcUa/freeopcua #304: Windows DLL build fails (no import library) + LGPL licensing

kiranpradeep, 2018-03-13:
I tried building this library on Windows 10 with Visual Studio. Static library build succeeds. But, the DLL build fails since no import libraries are generated. That is since there are no export symbols `(__declspec(dllexport))` in source code. So shared libraries cannot be built on Windows now. 

This library is licensed under LGPL and so users need to dynamically link to it.

cenit, 2019-03-22:
using [vcpkg](https://github.com/Microsoft/vcpkg) it should be possible to install freeopcua on windows and also mac/linux starting from sources, together with the most up-to-date boost libs. Please let me know if it works for you or any problem you may have, together we can fix it if there's any problem.

kiranpradeep, 2019-03-23:
@cenit The issue is straightforward. Unless we use `__declspec(dllexport)`, the dynamic libraries will not have exported functions. This is independent of vcpkg. An example is [[1](https://github.com/eclipse/mosquitto/blob/ae309b331cf79663d6a0f2df9a976c6511c552fa/lib/mosquitto.h#L26)]

cenit, 2019-03-23:
I did the port on vcpkg and there’s a significant patch set applied. I solved the `__declspec(dllexport)` using the power of cmake, that was the easiest part...
Again, please try it if you still need it, there might be some other problems but it is working on all operating systems, both statically and dynamically

kiranpradeep, 2019-03-23:
@cenit I don't use this library any more and so not planning on a pull request myself. I keep this issue open only so that anyone else who attempts to build this library in Windows (adhering to LGPL implying dynamic linking) could do so.

At the moment, the cmake trick is not in any branch. How does anyone else use it? There is no point we discussing this unless you plan to put a pull request which could close this issue. Or a pointer to documentation which could help build this as a dynamic library on Windows - might helps others in future. Kindly let me know if the expectation from me is any different.

cenit, 2019-03-23:
If you have vcpkg installed on your system, you can just do
```
vcpkg install freeopcua
```
And it will install the library on the system, whatever it is (being Linux, macOS, Windows), static or dynamic depending on your choice.
There is no need to open a pull request here, the patchset is already ported and accepted there ;)
I was just asking for feedback from users of the library. We plan to have the best stability, portability and ease of usage