FreeOpcUa/freeopcua #306: Standard cofiguration seems to be broken

bkuner, 2018-03-26:
When I start the example server it shows the error

[UaServer] [error] address_space_internal| parent node 'ns=0;i=11489;' does not exists

for the following list of nodes. Whats going wrong there? My last checkout of freeopcua was in 2016 - there it runs without problems ;-)

'ns=0;i=11489;'
'ns=0;i=11492;'
'ns=0;i=11580;'
'ns=0;i=11583;'
'ns=0;i=11585;'
'ns=0;i=11588;'
'ns=0;i=11590;'
'ns=0;i=11593;'
'ns=0;i=11629;'
'ns=0;i=11632;'
'ns=0;i=11634;'
'ns=0;i=11637;'
'ns=0;i=11639;'
'ns=0;i=11642;'
'ns=0;i=11680;'
'ns=0;i=11683;'
'ns=0;i=11685;'
'ns=0;i=11688;'
'ns=0;i=11690;'
'ns=0;i=11693;'
'ns=0;i=2949;' 
'ns=0;i=3875;' 
'ns=0;i=9029;' 
'ns=0;i=9069;' 
'ns=0;i=9111;' 
'ns=0;i=9113;' 
'ns=0;i=9213;'

ralf1070, 2018-03-27:
Older versions created a node structure without checking existence of nodes which were declared to be parent nodes. At one point it had become necessary to actually reference parent nodes and not only store the parent node id's, which led to these reports of missing parent nodes. As far as I know the actual code works as good as before - it just complains a bit more. Someone should fix the wrong references ...

JesseJ12345, 2019-04-28:
@bkuner , Did you fix this bug? How should I do ?   I still met this problem

bkuner, 2019-04-29:
No, I didn't. I use freeopcua as test server for my client so I can just ignore it.

JesseJ12345, 2019-04-29:
@bkuner ， thanks for your response, I will try another method to figure out, have a nice day !

bkuner, 2019-04-29:
If its just to figure out, what these numbers mean: 

ServerType_GetMonitoredItems,11489,Method
Server_GetMonitoredItems,11492,Method
FileType_Open,11580,Method
FileType_Close,11583,Method
FileType_Read,11585,Method
FileType_Write,11588,Method
FileType_GetPosition,11590,Method
FileType_SetPosition,11593,Method
NamespaceMetadataType_NamespaceFile_Open,11629,Method
NamespaceMetadataType_NamespaceFile_Close,11632,Method
NamespaceMetadataType_NamespaceFile_Read,11634,Method
NamespaceMetadataType_NamespaceFile_Write,11637,Method
NamespaceMetadataType_NamespaceFile_GetPosition,11639,Method
NamespaceMetadataType_NamespaceFile_SetPosition,11642,Method
ShelvedStateMachineType_TimedShelve,2949,Method
ConditionType_ConditionRefresh,3875,Method
CertificateGroupFolderType_DefaultHttpsGroup_TrustList_CloseAndUpdate_InputArguments,13875,Variable
ConditionType_AddComment,9029,Method
PublishedEventsType_DataSetWriterName_Placeholder_Diagnostics_Counters_StateDisabledByMethod,19029,Variable
DialogConditionType_Respond,9069,Method
AcknowledgeableConditionType_Acknowledge,9111,Method
PubSubConnectionType_WriterGroupName_Placeholder_Diagnostics_TotalInformation_Classification,19111,Variable
AcknowledgeableConditionType_Confirm,9113,Method
PubSubConnectionType_WriterGroupName_Placeholder_Diagnostics_TotalInformation_TimeFirstChange,19113,Variable
AlarmConditionType_ShelvingState_TimedShelve,9213,Method
PubSubConnectionType_ReaderGroupName_Placeholder_Diagnostics_Counters_StatePausedByParent_Classification,19213,Variable
But how to deal with? Implement just a node to get rid of the warning? To implement the methods seams to be hard work - and what is the benefit of it?

JesseJ12345, 2019-04-29:
@bkuner  ，I don't know how now but i will try to . I will try another opcua if i still can not solve it.  I think it's a bug not just a warning. THANKS A LOT.

eudoxos, 2019-06-11:
Any news on this one? I am hitting it with production code. Is there a workaround?