FreeOpcUa/freeopcua #313: C3861  identifier not found

Snarendranath, 2018-06-05:
Severity	Code	Description	Project	File	Line	Suppression State
Error	C3861	'IsZeroInt': identifier not found	server	c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.14.26428\include\opc\spdlog\fmt\bundled\format.cc	
('IsZeroInt','Precision Handler',' WidthHandler') showing not identified.

aniederm, 2018-08-22:
I have the same problem. is there any solution?
Thank you

deangelisdf, 2019-05-27:
Anyone find a solution?

ukeyice, 2019-08-06:
If you use visual studio, from the project property page, under c/c++ tab, change the conformance mode to No. This solves my problem.

HakanBayazitHabes, 2022-05-01:
You're referencing "functionThree( )" and "functionTwo( )" before they are even declared. The compiler reads a file from top to bottom. In function "functionOne( )", "functionTwo( )" was called, but the compiler had not been informed that "functionTwo( )" had existed.

To resolve this, you need to inform the compiler that "functionTwo( )" and "functionThree( )" exist before using them. For instance:

[image]