FreeOpcUa/freeopcua #315: Client sequence numbering

martinhaefner, 2018-06-12:
Hi all,

I am using your very smart library to communicate with a server written with the open62541 sdk. I get frequent disconnects from the server since the server only accepts messages from the client with strictly increasing sequence number. I had a look into the source of the freeopcua binray client and found a possible race condition. Fixing it, I have no further disconnects from the server.

There is a send_mutex in the binary_client.cpp and I put it around any sections where a sequence header is generated and the header is sent on the Stream later on, e.g.

    std::unique_lock<std::mutex> send_lock(send_mutex);

    const SequenceHeader sequence = CreateSequenceHeader();
    hdr.AddSize(RawSize(sequence));
    hdr.AddSize(RawSize(request));

    Stream << hdr << algorithmHeader << sequence << request << flush;

Maybe you find some time to fix this issue?! 

Regards,
Martin

ralf1070, 2018-06-13:
Is it possible for you to create a pull request containing this modification?

martinhaefner, 2018-06-14:
Done in #316 

Regards,
Martin