FreeOpcUa/freeopcua #318: How to build freeopcua by VisualStudio2015

mcdbcherry, 2018-08-15:
I have faced cmake error.
Could you kindly tell me correct way to build freeopcua with VS14?

*boost_1_56 was built with bellow comman. 
>bootstrap.bat
> b2 toolset=msvc threading=multi variant=debug,release link=static runtime-link=static address-model=32 --stagedir=stage/x86 -j 8
> b2 toolset=msvc threading=multi variant=debug,release link=shared runtime-link=shared address-model=32 --stagedir=stage/x86 -j 8
> b2 toolset=msvc threading=multi variant=debug,release link=static runtime-link=static address-model=64 --stagedir=stage/x64 -j 8
> b2 toolset=msvc threading=multi variant=debug,release link=shared runtime-link=shared address-model=64 --stagedir=stage/x64 -j 8

after build, boost_1_56 was placed C:\
C:\boost_1_56
+boost
+libs/accumulators ...
+stage/x64/lib/libboost_atomic-vc140-mt-s-1_56.lib ...

the command of build freeopcua as bellow.

>rem compiling
>mkdir build
>cd build
>cmake .. -DBOOST_ROOT=c:\boost_1_56

cmake output is bellow.

-- Selecting Windows SDK version  to target Windows 10.0.17134.
CMake Error at C:/Program Files/CMake/share/cmake-3.12/Modules/FindBoost.cmake:2044 (message):
  Unable to find the requested Boost libraries.

  Boost version: 1.56.0

  Boost include path: C:/boost_1_56

  Could not find the following Boost libraries:

          boost_system
          boost_program_options
          boost_filesystem
          boost_thread

  Some (but not all) of the required Boost libraries were found.  You may
  need to install these additional Boost libraries.  Alternatively, set
  BOOST_LIBRARYDIR to the directory containing Boost libraries or BOOST_ROOT
  to the location of Boost.
Call Stack (most recent call first):
  CMakeLists.txt:108 (find_package)

-- Boost INCLUDE DIR IS: C:/boost_1_56
-- Boost LIBRARY DIR IS:
-- SSL support using libmbedtls: ON
--   mbedtls LIBRARIES: mbedx509mbedtlsmbedcrypto
-- Compiling module for python 3.7.0
-- Could NOT find Boost
-- Boost python3 lib not found: Not building python3 module
-- Configuring incomplete, errors occurred!

Could you kindly tell me the process to build freeopcua on VisualStudio2015 ?

cenit, 2019-03-22:
using [vcpkg](https://github.com/Microsoft/vcpkg) it should be possible to install freeopcua on windows and also mac/linux starting from sources, together with the most up-to-date boost libs. Please let me know if it works for you or any problem you may have, together we can fix it if there's any problem.

sirop, 2020-05-04:
@cenit
 So if an option of freeopcua is OFF per default, then vcpkg will not support it?
For instance: https://github.com/FreeOpcUa/freeopcua/issues/288#issuecomment-348777470

cenit, 2020-05-04:
No! You can add a so called “feature” which will enable that option during build time :)

sirop, 2020-05-05:
@cenit 
Sorry, I did not find any example for "feature" option in the docs of vcpkg.
However, the master branch of freeopcua says that mbedtls is ON per default.
I will try to build now and report.

sirop, 2020-05-05:
@cenit No success until now:
```
PS D:\freeopcua\vcpkg> .\vcpkg.exe install freeopcua[mbedtls]:x64-windows --dry-run
Computing installation plan...
Package freeopcua does not have a mbedtls feature

  * boost-thread[core]:x64-windows
  * boost-throw-exception[core]:x64-windows
  * boost-tokenizer[core]:x64-windows
  * boost-tti[core]:x64-windows
  * boost-tuple[core]:x64-windows
  * boost-type-index[core]:x64-windows
  * boost-type-traits[core]:x64-windows
  * boost-typeof[core]:x64-windows
  * boost-unordered[core]:x64-windows
  * boost-utility[core]:x64-windows
  * boost-variant[core]:x64-windows
  * boost-vcpkg-helpers[core]:x64-windows
  * boost-winapi[core]:x64-windows
  * bzip2[core]:x64-windows
    freeopcua[core]:x64-windows
  * liblzma[core]:x64-windows
  * openssl[core]:x64-windows
  * openssl-windows[core]:x64-windows
  * zlib[core]:x64-windows
  * zstd[core]:x64-windows
Additional packages (*) will be modified to complete this operation.
PS D:\freeopcua\vcpkg>

```

cenit, 2020-05-06:
If you launch `.\vcpkg.exe install freeopcua:x64-windows --dry-run`, the freeopcua that you get is with or without mbedtls? If it is on by default, it should be on, otherwise please let me know so that we can fix it on vcpkg

sirop, 2020-05-06:
Beg your pardon for confusion: mbedtls is ON by default in the master branch of freeopcua,
but https://github.com/microsoft/vcpkg/blob/master/ports/freeopcua/portfile.cmake#L25 says it is OFF. Or does this option have a different meaning at this place?

Anyway, launching '.\vcpkg.exe install freeopcua:x64-windows' breaks off with the error message:
> -- Using cached D:/freeopcua/vcpkg/downloads/boostorg-regex-boost-1.72.0.tar.gz
> -- Using source at D:/freeopcua/vcpkg/buildtrees/boost-regex/src/ost-1.72.0-a233385537
> -- Building x64-windows-rel
> CMake Error at scripts/cmake/vcpkg_execute_required_process.cmake:72 (message):
>     Command failed: D:/freeopcua/vcpkg/installed/x64-windows/tools/boost-build/b2.exe --stagedir=D:/freeopcua/vcpkg/buildtrees/boost-regex/x64-windows-rel/stage --build-dir=D:/freeopcua/vcpkg/buildtrees/boost-regex/x64-windows-rel --user-config=D:/freeopcua/vcpkg/buildtrees/boost-regex/x64-windows-rel/user-config.jam -j2 --debug-configuration --debug-building --debug-generators --disable-icu --ignore-site-config --hash -q -sZLIB_INCLUDE=D:/freeopcua/vcpkg/installed/x64-windows/include -sBZIP2_INCLUDE=D:/freeopcua/vcpkg/installed/x64-windows/include -sLZMA_INCLUDE=D:/freeopcua/vcpkg/installed/x64-windows/include -sZSTD_INCLUDE=D:/freeopcua/vcpkg/installed/x64-windows/include threading=multi threadapi=win32 runtime-link=shared link=shared address-model=64 architecture=x86 toolset=msvc -sZLIB_BINARY=zlib -sZLIB_LIBPATH=D:/freeopcua/vcpkg/installed/x64-windows/lib -sBZIP2_BINARY=bz2 -sBZIP2_LIBPATH=D:/freeopcua/vcpkg/installed/x64-windows/lib -sLZMA_BINARY=lzma -sLZMA_LIBPATH=D:/freeopcua/vcpkg/installed/x64-windows/lib -sZSTD_BINARY=zstd -sZSTD_LIBPATH=D:/freeopcua/vcpkg/installed/x64-windows/lib variant=release debug-symbols=on
>     Working Directory: D:/freeopcua/vcpkg/buildtrees/boost-regex/src/ost-1.72.0-a233385537/build
>     Error code: 1
>     See logs for more information:
>       D:\freeopcua\vcpkg\buildtrees\boost-regex\build-x64-windows-rel-out.log
> 
> Call Stack (most recent call first):
>   installed/x64-windows/share/boost-build/boost-modular-build.cmake:302 (vcpkg_execute_required_process)
>   ports/boost-regex/portfile.cmake:20 (boost_modular_build)
>   scripts/ports.cmake:90 (include)
> 
> 
> Error: Building package boost-regex:x64-windows failed with: BUILD_FAILED
> Please ensure you're using the latest portfiles with `.\vcpkg update`, then
> submit an issue at https://github.com/Microsoft/vcpkg/issues including:
>   Package: boost-regex:x64-windows
>   Vcpkg version: 2020.02.04-nohash

The contents of file:///D:/freeopcua/vcpkg/buildtrees/boost-regex/build-x64-windows-rel-out.log can be found in https://gist.github.com/sirop/7aa0a765f6f002faac9e710a1b0d0948 .