FreeOpcUa/freeopcua #325: terminate called after throwing an instance of 'std::runtime_error'   what():  Response timed out

sschakravarty, 2019-02-21:


sschakravarty, 2019-02-21:
I am using subscription of a tag - the application is running fine and there is response timed out exception aborting my application.

Can any body say - why abort() is being executed 
and how to solve the issue?

sschakravarty, 2019-02-21:
Subscription call back exception --> 
terminate called after throwing an instance of 'std::runtime_error'
  what():  Response timed out
Aborted

please find for above comment

strowk, 2019-03-08:
By the way, this problem reproduces also on example server built from `master`, while does not reproduced in `stable` branch.

Here's client's output for stable:
```
Connecting to: opc.tcp://127.0.0.1:4841/freeopcua/server/
Root node is: Node(QualifiedName(0:Root), ns=0;i=84;)
Child of objects node are: 
    Node(QualifiedName(0:Server), ns=0;i=2253;)
    Node(QualifiedName(2:NewObject), ns=2;i=99;)
NamespaceArray is: 
    http://opcfoundation.org/UA/
    http://freeopcua.github.io
    http://examples.freeopcua.github.io
got node: Node(QualifiedName(2:MyVariable), ns=2;i=1001;)
Got sub handle: 101, sleeping 5 sconds
Received DataChange event, value of Node Node(QualifiedName(2:MyVariable), ns=2;i=1001;) is now: (Print of arbitrary variant not suported yet)
Received DataChange event, value of Node Node(QualifiedName(2:MyVariable), ns=2;i=1001;) is now: (Print of arbitrary variant not suported yet)
Disconnecting
```

And the same for master:

```
[2019-03-08 12:28:46.125] [client] [info] Connecting to: opc.tcp://127.0.0.1:4840/freeopcua/server/
Failed to close socket connection. Transport endpoint is not connected
[2019-03-08 12:28:46.176] [client] [info] ua_client             | creating session ...
[2019-03-08 12:28:46.179] [client] [info] ua_client             | create session OK
[2019-03-08 12:28:46.179] [client] [info] ua_client             | activating session ...
[2019-03-08 12:28:56.191] [client] [info] ua_client             | CloseSession response is 
Failed to close socket connection. Transport endpoint is not connected
[2019-03-08 12:28:56.191] [client] [error] Error: Response timed out

```
It seems that something is broken in master.
I'm going to use the `stable`  for now, does anybody knows how is the state of it?

cnjlq84, 2021-01-22:
anybody knows how to solve this issue?