FreeOpcUa/freeopcua #326: mbedtls/entropy.h: No such file

strowk, 2019-03-08:
Hi!

I used `debian.soft` from 
https://github.com/FreeOpcUa/freeopcua#ubuntu-debians
and then cmake build
https://github.com/FreeOpcUa/freeopcua#linux
and on `make` it ended with
```
/home/tim/shr/freeopcua/src/client/client.cpp:29:29: fatal error: mbedtls/entropy.h: No such file or directory
```
, then I tried to do
```
sudo apt-get install libmbedtls-dev
```
and `make` again and it was success. 

So I guess that `libmbedtls-dev` should be included in the debian.soft. 

Now I see this issue 
https://github.com/FreeOpcUa/freeopcua/issues/307
and recommendations provided, but let's be honest - it is a workaround. 

Users who follow step-by-step instructions should be able to get the result independently from their initial system state, especially if the state (as for my case) was the initial state of newly installed Debian. 

I mean if debian.soft is provided for Debian users - either it should include all dependencies or `SSL_SUPPORT_MBEDTLS` should be disabled by default.

Other way could be include recommendation in the docs, but I'm sure that either installation of mbedtls or disabling of SSL_SUPPORT_MBEDTLS by default is a better solution.

mxbuschi, 2020-01-22:
Thank you, the same issue arose on my system, installing `libmbedtls-dev` solved the issue.

zulfikar4568, 2020-12-04:
Thank you this solved my issue!

destogl, 2021-01-07:
This should be added into `cmake ..` checks before compile...