FreeOpcUa/freeopcua #338: Visual Studio 2019 C1001: Internal Error has occurred

smedellin90, 2019-07-30:
The project was built on my system using boost_1_60 and cmake version 3.14.0. The prompt printed out that the build was successful, and I am able to open the project on Visual Studio 2019.

When I go to compile `example_client.cpp` , I get the following compiler error:

`1>------ Build started: Project: example_server, Configuration: Debug x64 ------
1>example_server.cpp
1>Unknown compiler version - please run the configure tests and report the results
1>C:\Users\smedellin\Documents\opcua\freeopcua\include\opc/ua/protocol/variant.h(87,1): error C1001:  An internal error has occurred in the compiler.
1>C:\Users\smedellin\Documents\opcua\freeopcua\include\opc/ua/protocol/variant.h(87,1): error C1001: (compiler file 'd:\agent\_work\3\s\src\vctools\Compiler\CxxFE\sl\p1\c\types.c', line 4589)
1>C:\Users\smedellin\Documents\opcua\freeopcua\include\opc/ua/protocol/variant.h(87,1): error C1001:  To work around this problem, try simplifying or changing the program near the locations listed above.
1>C:\Users\smedellin\Documents\opcua\freeopcua\include\opc/ua/protocol/variant.h(87,1): error C1001: Please choose the Technical Support command on the Visual C++
1>C:\Users\smedellin\Documents\opcua\freeopcua\include\opc/ua/protocol/variant.h(87,1): error C1001:  Help menu, or open the Technical Support help file for more information
1>C:\Users\smedellin\Documents\opcua\freeopcua\include\opc/ua/protocol/variant.h(93): message :  see reference to class template instantiation 'OpcUa::has_begin_end<T>' being compiled
1>Done building project "example_server.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========`

I have tried to fix by adjusting the last if statement of `visualc.hpp`  to

if  (MSC_VER > 1922)

instead of 

if  (MSC_VER > 1900)

Nevertheless, I still get the error listed above.

AdmiralPellaeon, 2019-08-21:
I also have problems with the variant. Can't compile the library anymore:

2>D:\projects\cpp\freeopcua-master-2019-08-21\include\opc/ua/protocol/variant.h(87,1): error C1001:  Interner Compilerfehler.

614ir, 2020-05-11:
> The project was built on my system using boost_1_60 and cmake version 3.14.0. The prompt printed out that the build was successful, and I am able to open the project on Visual Studio 2019.
>
> When I [...]

So, did you solve it? I am having exactly the same problem :/

AdmiralPellaeon, 2020-05-11:
Yes, I managed to compile it, but that's some months ago. As far as i remember I didn't change the code. Perhaps you can try it with a newer version of boost.

614ir, 2020-05-11:
Okey, thank you for the response

AdmiralPellaeon, 2020-05-11:
I checked my files. Attached you find all src files with havbe a newer timne stamp compared to the check out solution from git.
And I definetly use much 
[changed_files.zip](https://github.com/FreeOpcUa/freeopcua/files/4608523/changed_files.zip)
newer version of boost.