FreeOpcUa/freeopcua #352: spdlog conflict with ros2 logging

nicholas477, 2020-01-28:
# Summary
I am dynamically linking against both freeopcua and [rclcpp](https://github.com/ros2/rclcpp) and they both use differing versions of spdlog and I think it is causing a segfault in my program. When I run my program in the debugger, it segfaults when I init the ros2 runtime here: https://github.com/ros2/rcl_logging/blob/fc3a6f37ada01d21242eb146d1f7715a1f1e6a00/rcl_logging_spdlog/src/rcl_logging_spdlog.cpp#L145
but the segfault doesn't happen when I don't link against freeopcua.

Right now, to get my program working, i'm using the log4cxx logger in ros instead of the spdlog logger, but I think this spdlog issue is something that should be solved in freeopcua. It looks like the ros logger [links against the system's version of spdlog](https://github.com/ros2/rcl_logging/blob/fc3a6f37ada01d21242eb146d1f7715a1f1e6a00/rcl_logging_spdlog/CMakeLists.txt#L17), and I think it should be the same in freeopcua, and although I haven't tested it, I assume this would fix my issue I'm having with both libraries.

# Possible Fixes
Use (or add the option to use) the system version of spdlog in cmake.

apoorvcn47, 2020-09-25:
I am new here, @morxa Can you please explain how i can compile against system spdlog? I am having the same issue as #352. I believe your commits are not merged yet.

morxa, 2020-09-26:
Best solution would be to either get them to merge the fix or merge it yourself!

apoorvcn47, 2020-09-26:
I made the changes on my side. i hope they accept your merge

PR #356 (2020-11-23): Use system spdlog if available
Check for a system-wide installation of spdlog and use it if found.

Commit fd2e53aeccb03a6ce9c231735615ec2b3cd08742 is necessary because logging the log level fails with more recent versions of spdlog. I have not investigated why.

This should also fix #352.
closed 2020-11-23 by destogl