FreeOpcUa/freeopcua #357: VS2013, CMake 3.17,Compile error

18366102135, 2020-04-23:
Sorry, just contacted C + +, I don't understand what's wrong, please help me to have a look, thank you!

[image]

tyronedong, 2020-05-13:
same problem, have anyone solved this?

plastictown, 2020-07-08:
> same problem, have anyone solved this?

Manually fixed it: replace the function signature with Initialize (const std :: string & uriString), and instead of size use uriString.size( ), and uriString.c_str( ) in places where const char is needed

destogl, 2020-07-09:
HI @plastictown, this is great, can you open a Pull request?

TrigoldenZx, 2020-11-24:
i hava the same problem, how to fix it?

18366102135, 2020-11-24:
I used the previous version，It can also be modified in this way： replace the function signature with Initialize (const std :: string &amp; uriString), and instead of size use uriString.size( ), and uriString.c_str( ) in places where const char is needed。

TrigoldenZx, 2020-11-28:
> 
> I used the previous version，It can also be modified in this way： replace the function signature with Initialize (const std :: string &amp; uriString), and instead of size use uriString.size( ), and uriString.c_str( ) in places where const char is needed。
> […](#)

thank you for your help, i have fixed the problem about uriString,but there is the other error :error C3861:  "close" : can't find the symbol, did you fix this?

plastictown, 2020-11-28:
> thank you for your help, i have fixed the problem about uriString,but there is the other error :error C3861: "close" : can't find the symbol, did you fix this?

if I'm not mistaken, you should use the closesocket() function instead of close()

plastictown, 2020-11-28:
> HI @plastictown, this is great, can you open a Pull request?

I'm afraid this fix may affect the implementation for different platforms. I think the code authors can make alternative implementations for C and c++, as well as for wsapi and the POSIX socket interface.