FreeOpcUa/freeopcua #362: [server] [error] address_space_internal| parent node xxxxxx does not exists

kamarya, 2020-07-14:
I am on master branch (see below) and the server initialization fails
```
commit d24d7806a71739b869f6d2465316184b49e37875 (HEAD -> master, origin/master, origin/HEAD)
Merge: f7c41d2 0310faf
Author: Denis Štogl <destogl@users.noreply.github.com>
Date:   Wed Jun 17 10:52:00 2020 +0200

    Merge pull request #361 from morxa/boost-173-std-algorithm-includes
    
    Add missing includes to algorithm header

```

Here are the errors

```
[2020-07-14 10:07:14.987] [server] [error] address_space_internal| parent node 'ns=0;i=11489;' does not exists
[2020-07-14 10:07:14.987] [server] [error] address_space_internal| parent node 'ns=0;i=11489;' does not exists
[2020-07-14 10:07:15.009] [server] [error] address_space_internal| parent node 'ns=0;i=11580;' does not exists
[2020-07-14 10:07:15.010] [server] [error] address_space_internal| parent node 'ns=0;i=11580;' does not exists
[2020-07-14 10:07:15.010] [server] [error] address_space_internal| parent node 'ns=0;i=11583;' does not exists
[2020-07-14 10:07:15.010] [server] [error] address_space_internal| parent node 'ns=0;i=11585;' does not exists
[2020-07-14 10:07:15.010] [server] [error] address_space_internal| parent node 'ns=0;i=11585;' does not exists
[2020-07-14 10:07:15.010] [server] [error] address_space_internal| parent node 'ns=0;i=11588;' does not exists
[2020-07-14 10:07:15.010] [server] [error] address_space_internal| parent node 'ns=0;i=11590;' does not exists
[2020-07-14 10:07:15.010] [server] [error] address_space_internal| parent node 'ns=0;i=11590;' does not exists
[2020-07-14 10:07:15.010] [server] [error] address_space_internal| parent node 'ns=0;i=11593;' does not exists
[2020-07-14 10:07:15.012] [server] [error] address_space_internal| parent node 'ns=0;i=11629;' does not exists
[2020-07-14 10:07:15.012] [server] [error] address_space_internal| parent node 'ns=0;i=11629;' does not exists
[2020-07-14 10:07:15.012] [server] [error] address_space_internal| parent node 'ns=0;i=11632;' does not exists
[2020-07-14 10:07:15.012] [server] [error] address_space_internal| parent node 'ns=0;i=11634;' does not exists
[2020-07-14 10:07:15.012] [server] [error] address_space_internal| parent node 'ns=0;i=11634;' does not exists
[2020-07-14 10:07:15.012] [server] [error] address_space_internal| parent node 'ns=0;i=11637;' does not exists
[2020-07-14 10:07:15.012] [server] [error] address_space_internal| parent node 'ns=0;i=11639;' does not exists
[2020-07-14 10:07:15.012] [server] [error] address_space_internal| parent node 'ns=0;i=11639;' does not exists
[2020-07-14 10:07:15.012] [server] [error] address_space_internal| parent node 'ns=0;i=11642;' does not exists
[2020-07-14 10:07:15.014] [server] [error] address_space_internal| parent node 'ns=0;i=11680;' does not exists
[2020-07-14 10:07:15.014] [server] [error] address_space_internal| parent node 'ns=0;i=11680;' does not exists
[2020-07-14 10:07:15.014] [server] [error] address_space_internal| parent node 'ns=0;i=11683;' does not exists
[2020-07-14 10:07:15.014] [server] [error] address_space_internal| parent node 'ns=0;i=11685;' does not exists
[2020-07-14 10:07:15.014] [server] [error] address_space_internal| parent node 'ns=0;i=11685;' does not exists
[2020-07-14 10:07:15.014] [server] [error] address_space_internal| parent node 'ns=0;i=11688;' does not exists
[2020-07-14 10:07:15.014] [server] [error] address_space_internal| parent node 'ns=0;i=11690;' does not exists
[2020-07-14 10:07:15.014] [server] [error] address_space_internal| parent node 'ns=0;i=11690;' does not exists
[2020-07-14 10:07:15.014] [server] [error] address_space_internal| parent node 'ns=0;i=11693;' does not exists
[2020-07-14 10:07:15.048] [server] [error] address_space_internal| parent node 'ns=0;i=11492;' does not exists
[2020-07-14 10:07:15.048] [server] [error] address_space_internal| parent node 'ns=0;i=11492;' does not exists
[2020-07-14 10:07:15.077] [server] [error] address_space_internal| parent node 'ns=0;i=9029;' does not exists
[2020-07-14 10:07:15.077] [server] [error] address_space_internal| parent node 'ns=0;i=3875;' does not exists
[2020-07-14 10:07:15.079] [server] [error] address_space_internal| parent node 'ns=0;i=9069;' does not exists
[2020-07-14 10:07:15.080] [server] [error] address_space_internal| parent node 'ns=0;i=9111;' does not exists
[2020-07-14 10:07:15.080] [server] [error] address_space_internal| parent node 'ns=0;i=9113;' does not exists
[2020-07-14 10:07:15.082] [server] [error] address_space_internal| parent node 'ns=0;i=9213;' does not exists
[2020-07-14 10:07:15.085] [server] [error] address_space_internal| parent node 'ns=0;i=2949;' does not exists
```

destogl, 2020-07-14:
Hi @kamarya,

who are you exactly starting the server?

kamarya, 2020-07-17:
who or how !? non-root and just simply run `./example_server`

swamper123, 2020-07-17:
Are these semicolons at the end of the node IDs not to much? This may cause errors, because ";" is not a number.

treww, 2020-07-17:
https://reference.opcfoundation.org/v104/Core/docs/Part6/5.3.1/#5.3.1.10

swamper123, 2020-07-20:
So in case of an ExpandedNodeId another `;` is possible, but would still need a postfix of` <type>=<value>` [. Source here.](https://reference.opcfoundation.org/v104/Core/docs/Part6/5.3.1/#5.3.1.11).

destogl, 2020-07-20:
I will test it tomorrow morning (CET)

destogl, 2020-07-21:
@kamarya @swamper123 I get also the same error...

We should check what is happening. I will not manage to do this soon... Until then, the server is still at least working, there are some errors when using UaExpert, but it is OK for the testing.

swamper123, 2020-07-21:
@kamarya Are these the only errors? Can you debug where and how this parent node is created?

kamarya, 2020-07-21:
@destogl @swamper123 for now I am evaluating this library. I do not think that I will have time to do more about this. I was hoping that the principal contributors take it seriously and fix it. It does not seem to be complicated.

FelixCrazzolaraPick8Ship, 2023-02-18:
For everyone encountering this issue, you can fix it by merging the code in `schemas/generate_address_space.py` from `freeopcua/python-asyncio` for `UAMethod` and then regenerate the files `standard_address_space_part5.cpp` and `standard_address_space_part9.cpp`.

issam21ks, 2024-03-31:
@FelixCrazzolaraPick8Ship Hello, I'm having the same issue, but have no idea how to merge the code for UAMethod and regenerate the files. Can you please elaborate in more detail? I would really appreciate it.

guojing555, 2024-03-31:
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

FelixCrazzolaraPick8Ship, 2024-04-02:
See [here](https://github.com/FelixCrazzolaraPick8Ship/freeopcua).