FreeOpcUa/freeopcua #376: Error on build example

darlandieterich, 2020-11-11:
Hi!

When run the command:
`g++ example_client.cpp -lopcuacore -lopcuaclient -lopcuaprotocol -lpthread -lmbedtls -lmbedcrypto -lmbedx509 -o client`

has this errors:
```
/usr/bin/ld: could not be found -lopcuacore
/usr/bin/ld: could not be found -lopcuaclient
/usr/bin/ld: could not be found -lopcuaprotocol
collect2: error: ld returned 1 exit status
```

Any help?

pbertoni89, 2020-11-11:
Examples should build smoothly with a

```
mkdir build && cd $_
cmake -DBUILD_SERVER=on -DBUILD_CLIENT=on -DBUILD_PYTHON=off -DBUILD_TESTING=off ..
make -j
```

Did you tried?

darlandieterich, 2020-11-12:
I tried and the error message appears

ljonka, 2022-09-29:
try "make clean" before and repeat

guojing555, 2022-09-29:
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。