FreeOpcUa/freeopcua #380: Import xml error

zx-zx-zx, 2021-01-12:
Question 1：The following error occurred while adding the XML file（Opc.Ua.PlasticsRubber.GeneralTypes.NodeSet2.xml）：opcua.ua.uaerrors._auto.BadNodeIdExists: "The requested node id is already used by another node."(BadNodeIdExists)
cryptography is not installed, use of crypto disabled
cryptography is not installed, use of crypto disabled
uamodeler.model_manager - INFO - Starting server on opc.tcp://0.0.0.0:48400/freeopcua/uamodeler/')
uamodeler.server_manager - INFO - Starting python-opcua server')
opcua.server.server - WARNING - Endpoints other than open requested but private key and certificate are not set.')
opcua.server.binary_server_asyncio - WARNING - Listening on 0.0.0.0:48400')
IMPORT d:/我的文档/桌面/zxiang/OPC UA/open62541/deps/ua-nodeset/DI/Opc.Ua.Di.NodeSet2.xml
opcua.common.xmlparser - WARNING - Parsing value of type 'QualifiedName' not implemented')
IMPORT d:/我的文档/桌面/zxiang/OPC UA/open62541/deps/ua-nodeset/PlasticsRubber/GeneralTypes/1.02/Opc.Ua.PlasticsRubber.GeneralTypes.NodeSet2.xml
opcua.common.xmlimporter - WARNING - failure adding node NodeData(nodeid:NumericNodeId(ns=1;i=6318))')
uawidgets.utils - ERROR - ('Error no extension class registered ', 'PIDParametersDataType', NumericNodeId(ns=2;i=3023))')
Traceback (most recent call last):
  File "C:\Users\admin\AppData\Roaming\Python\Python39\site-packages\uawidgets\utils.py", line 21, in wrapper
    result = func(self, *args)
  File "C:\Users\admin\AppData\Roaming\Python\Python39\site-packages\uamodeler\refnodesets_widget.py", line 42, in add_nodeset
    self.import_nodeset(path)
  File "C:\Users\admin\AppData\Roaming\Python\Python39\site-packages\uamodeler\refnodesets_widget.py", line 50, in import_nodeset
    self.server_mgr.import_xml(path)
  File "C:\Users\admin\AppData\Roaming\Python\Python39\site-packages\uamodeler\server_manager.py", line 68, in import_xml
    return self._backend.import_xml(path)
  File "C:\Users\admin\AppData\Roaming\Python\Python39\site-packages\uamodeler\server_manager.py", line 112, in import_xml
    return self._server.import_xml(path)
  File "C:\Users\admin\AppData\Roaming\Python\Python39\site-packages\opcua\server\server.py", line 510, in import_xml
    return importer.import_xml(path, xmlstring)
  File "C:\Users\admin\AppData\Roaming\Python\Python39\site-packages\opcua\common\xmlimporter.py", line 67, in import_xml
    node = self._add_node_data(nodedata)
  File "C:\Users\admin\AppData\Roaming\Python\Python39\site-packages\opcua\common\xmlimporter.py", line 86, in _add_node_data
    node = self.add_variable(nodedata)
  File "C:\Users\admin\AppData\Roaming\Python\Python39\site-packages\opcua\common\xmlimporter.py", line 207, in add_variable
    attrs.Value = self._add_variable_value(obj,)
  File "C:\Users\admin\AppData\Roaming\Python\Python39\site-packages\opcua\common\xmlimporter.py", line 291, in _add_variable_value
    extobj = self._make_ext_obj(ext)
  File "C:\Users\admin\AppData\Roaming\Python\Python39\site-packages\opcua\common\xmlimporter.py", line 238, in _make_ext_obj
    ext = self._get_ext_class(obj.objname)()
  File "C:\Users\admin\AppData\Roaming\Python\Python39\site-packages\opcua\common\xmlimporter.py", line 233, in _get_ext_class
    raise Exception("Error no extension class registered ", name, nodeid)
Exception: ('Error no extension class registered ', 'PIDParametersDataType', NumericNodeId(ns=2;i=3023))

Question 2：Can I export. BSD file，i need Opc.Ua.PlasticsRubber.GeneralTypes.Types.bsd

destogl, 2021-01-13:
Hi @zx-zx-zx,

Q1: you are using some node ID which is already used by the system: Opc.Ua.PlasticsRubber.GeneralTypes.NodeSet2.xml）：opcua.ua.uaerrors._auto.BadNodeIdExists: "The requested node id is already used by another node."(BadNodeIdExists)
Solution: If you can, change your NodeId by e.g. change namespace

Q2: I don't think so. Try  FreeOpcUa/opcua-modeler. the tool is created for such things.

zx-zx-zx, 2021-01-14:
> Hi @zx-zx-zx,
>
> Q1: you are using some node ID which is already used by the system: Opc.Ua.PlasticsRubber.GeneralTypes.NodeSet2.xml）：opcua.ua.uaerrors._auto.BadNodeIdExists: "The requested node id is [...]

It's a standard specification document, and I can't change.

> Q2: I don't think so. Try FreeOpcUa/opcua-modeler. the tool is created for such things.

But I didn't find the relevant function, can you tell me how to do it